micro-nova/AmpliPi #401: Internet Radio station silent at re-launch

rickballard, 2022-11-11:
Latest software... When touching the screen to activate a keypad (have 2 keypads), the internet radio station shows as visible from last use.  However, touching the mute button and or the play button, will not start the station playing.  To do this, one must change station to another station and then change the station back again.  Have tried waiting several minutes for the original station to connect, and have tried moving the volume slider, but the only fix seems to be changing the station.  It is as if the cache memory of the Internet Radio station URL gets corrupted, if the music is left off for a while.

linknum23, 2022-11-14:
Sorry about your troubles Rick.

It sounds like the internet radio station is failing after a while. We have added some support in AmpliPi to handle failures and restart the radio station, but a small portion of radio stations in our experience still have some issues and need to be restarted after a while. This is likely what is happening here, and why switching the station is necessary to recover it.

Fixing this will be simpler in the future: In our latest touch panel [release](https://github.com/micro-nova/AmpliPiWallPanel/releases/tag/0.2.7), you should just need to hit play on a failed radio station to start playing again.

I hope this helps.

rickballard, 2022-11-14:
That is exactly what I was hoping for.  However, you could go one better and utilize an URL listing service for Internet radio stations.  The URLs are not trivial to figure out, as folks have to inspect the right DIV in the live streaming website of the station using browser-native developer tools.  And the URLs change each year.

linknum23, 2022-11-14:
We do need to do a third pass on internet radio stations, to make them easier to use.

Have you tried using **Search Stations** when creating an internet radio stream? Behind the scenes it uses https://www.radio-browser.info/search to find internet radio stations. 

That should make it easy to find a large selection of internet radio stations. It would at least avoid the inspect element problem you are having.

rickballard, 2022-11-19:
https://www.radio-browser.info/search ....  this link is excellent, and should be in the setup menu whenever folks are trying to  add streaming stations.  You might also consider automating the population of configuration fields, for each station folks select to add to their station list.  That way, you could ensure that if the station URL changes, folks won't need to worry about updating links, etc, because the auto-populate function would take care of all that for them.

linknum23, 2023-01-30:
Closing this for now. We will add a reminder to continue polish up our internet radio streaming interface.

closed 2023-01-30 by linknum23