micro-nova/AmpliPi #402: Hiss/chatter on ALL sources from RTL-SDR-USB-FM-dongle

rickballard, 2022-11-11:
Likely a hardware bug: 

There appears to be inadequate signal isolation on the input of ALL sources, from the signal input by the Micro-Nova-suggested USB FM-antenna USB input dongle.  

Input signals that have been amplified by a Channel Master antenna amp, or signal aggregator, then fed into the "RTL-SDR Blog R820T2 RTL2832U 1PPM TCXO SMA Software Defined Radio dongle", will always create a background hiss, with a chatter much like that of old fashioned dial-up modems.  

This is most noticeable while an internet radio station, or other sources, is at  low volume.  The background hiss/chatter seems to remain at a fairly constant volume, even when the internet radio station volume is adjusted.  It can even be heard when the internet radio station is muted.

This background hiss/chatter can lead to extreme human suffering and mental retardation, as it seems to be "tuned" to create maximum insanity when otherwise trying to concentrate on one's regular day job.  Micro-Nova could thus generate an additional revenue stream re-branding this feature as a covert torture-device, for use on, say, Twitter employees who are not working hard enough...?

concorde-7, 2023-01-15:
@rickballard have you gotten any resolution to this issue?

rickballard, 2023-01-16:
A simple FM antenna and a large ATSC-video antenna, are both connected to an RF distribution amplifier,which feeds the RTL-SDR USB-dongle for the AmpliPi.  The combination of these antenna signals, and/or their amplification, may cause an overload in the AmpliPi, especially if the AmplifPi has limited or zero protection against strong RF input signals.  If I instead, directly connect the FM antenna to the USB dongle, the background hissing/buzzing noise may reduce.  I will try this later this week and let you know.  Thanks.

concorde-7, 2023-01-16:
If you still have trouble with the noise (and you aren't getting good support from the manufacturer) something else to try is to lend the unit to [Audio Science Review](https://www.audiosciencereview.com/forum/index.php?threads%2Fwe-measure-audio-equipment-for-free.1995%2F) for them to take a close look at. They review amps for free, and they do a super thorough job ([here is a recent review for an example](https://www.audiosciencereview.com/forum/index.php?threads/hypex-ncx500-class-d-amplifier-review.41007/)). I am sure they can get to the bottom of any noise issues you're hearing.

rickballard, 2023-01-16:
MicroNova have been excellent in the customer support department.  It is my own tardiness that has delayed resolution of this issue, for which I apologize profusely.  I will work on it later this week.  Meanwhile, if any other AmpliPi owner, or MicroNova themselves, wishes to submit a unit to professional independent audiophile-reviewers, I would be most interested in the advice and insights that result.  If you are a such a reviewer yourself, may I suggest you send a proposal to MicroNova, that comits you to at least a minimum suite of tests and return (or discount purchase) of the unit, depending upon whether you wish to keep it.  The box is open source, so it seems quite in keeping with its design philosophy to encourage crowdsourced reviews, especially if you also use professional methodologies.  Cheers.

concorde-7, 2023-01-17:
OK that is encouraging that support has been good. I'm considering AmpliPi but am concerned about noise issues like yours (and the lack of any independent technical reviews.)  I'll stay subscribed to this issue, hoping to see it resolved for you.

rickballard, 2023-01-17:
The product is still raw, needs refining with further software updates, but
it is an audiophile quality amp.  Much depends on your speakers so YMMV.

The keypad interface is not particularly pretty or colourful, a bit
utilitarian even, and the hard buttons are still unused unless you access
them via the AmpliPi's open API.  There is a boatload of Sonoff devices and
apps that could be added to AmpliPi, if the two companies were not mutually
exclusive in the proprietary versus open-API department.

The AmpliPi mobile app is basically an ugly web page, built with json, and
it sorely lacks an always present indicator to show which room/zone is
currently being displayed.  The entire idea of sources is still being
evaluated, they seem to be a distraction in general use.

There is no clock/alarm/scheduling features, built in, yet.  It needs a
reset pin to wake it up, after a power cut.  Programming the Internet Radio
Station URLs is not intuitive or automated, and may need re-doing if URLs
change.  Spotify works well as a source but it can be a bit fiddly to get
going as well.  Definitely still a Nerd's product, certainly not something
my wife would want to set up.

However, for an open source project, it is truly a remarkable effort, that
has serious potential if they keep at it.  Especially if you also use the
AmpliPi box, or a same network server, to host Home Assistant.  I enjoy
using it and it just keeps getting better every month.

Meanwhile, you can become part of the development effort quite easily if
you want to help out.  Your audiophile testing skills would likely be much
appreciated.

Regards, Rick.  416.414.8100

Lohrer, 2023-01-17:
I know @linknum23 and @kjk2010 did the primary work with the RTL-SDR in PR #151, but I'm interested to see what you find in your testing Rick. We know that joining the analog and digital power domains - for example plugging a single device into both the USB port and the 3.5mm aux input - will add noise to the audio. To my knowledge the RTL-SDR just uses the USB port so there _shouldn't_ be any issue from that.

Just to make sure I'm understanding 100%, you're saying an **Internet Radio** stream, not **FM Radio**, has noise when your RTL-SDR is plugged in, but if you remove the RTL-SDR the noise goes away? To help narrow things down, I'd also recommend removing anything else connected to AmpliPi during your testing. For example is the "Channel Master antenna amp" connected to AmpliPi in any way?

rickballard, 2023-01-17:
Michael, all sources, including Internet Radio URLs, have this noise from the RTL-SDR.  It is a very distinctive noise, designed by the Gremlins, to maximize the suffering of those who hear it.  It is like an FM-radio hiss, with a 1990's dialup modem inserting bursts of lower frequency buzz-chatter, with seemingly random durations and at seemingly random intervals.

I am no expert, but I suspect this buzz+hiss to be an analogue+digital hybrid, which gets through the filter circuitry via inductive cross-talk along the tracks of the motherboard itself.  The board likely has a lot of long parallel paths, or even a loop or two.  

The buzz may be an analogue facsimili, using the analogue hiss as a carrier.  Ff the buzz was truly digital, it would be more likely to interfere with control circuit operation, and the analogue hiss amplitude would remain constant.  Yet it seems to increase when the buzz occurs.  

Even if this strange noise problem proves to be caused by an over-driven input to the third-party RTL-SDR, it should still not happen.  If the AmpliPi's DAC was able to monitor noise on the AmpliPi's inputs, somewhat directly, this feature could be used for troubleshooting AND for automated input detection.  Also, noise can be used to pass signalling back and forth between smart speakers in the home, which could then be used for presence detection in home security ... or for voice control of AmpliPi without having to use Google's API's/licenses ... or for torturing folks...  

Did I mention that this buzzy-hiss noise was designed by the Gremlins to maximize suffering...?

gorski123, 2023-01-18:
Hey @rickballard, I just wanted to chime in here and piggy back on what Michael previously mentioned:

The Raspberry Pi in the AmpliPi can be noisy and as such we have completely isolated it from the analog electronics. Inside the AmpliPi we have a digital ground and an analog ground. The PI and digital electronics live on the digital side of the system. The DACs, analog front end electronics, etc, live on the analog side of the system. They are electrically isolated from one-another.

On the back of the unit, some ports live on the digital side of the system, and some live on the analog side.
For example, the USB ports are connected to the digital ground and the RCA jacks are connected to the analog ground. If you were to plug a single device into both of those ports, it could potentially bridge grounds together, which will 'release the Gremlins' and potentially cause the symptoms you are describing. 

We want to duplicate your setup here with the RTL-SDR here and see if we can help debug your issue. In order to do this, we need to know exactly what you have connected to the system when you are doing your tests. If you could snap a picture of the back of the unit with your connections and send it to us, that would be ideal. 

I sound mention here that this is just a theory, there could be something completely unrelated causing the Gremlins, but I think this is a good place to start looking.