micro-nova/AmpliPi #411: Feature Request: Clock/Alarm

rickballard, 2022-12-02:
We turn the AmpliPi on at the same time every weekday.  If it could turn itself on, it would be a pleasant alarm, (perhaps even ensuring that one fully wakes up, to turn down the volume on the wall keypad or phone app). 

Please could you consider adding a time/date field to manage current time/date, and to scheduled event conditions.  This feature would greatly enhance the number of use cases available and hence the value of this premium product.  

Alternatively, could you post a cheap app in the Play Store to link the native Android clock/alarm, with AmpliPi's per-zone playback controls?  Or am I missing a user-friendly method that already exists...?

linknum23, 2022-12-02:
This is something we would definitely consider adding if there are enough requests for it. 

In the meantime something like this can be done relatively simply with a combination of node red and the presets in our api.  We realize that some of this stuff can be a little intimidating to work with though. Here is a walk through for making an alarm with node red and AmpliPi. 

First lets make an alarm preset using amplipis api here: http://amplipi.local/doc#post-/api/preset

Edit the create_preset example to have the following:

```json
{
  "name": "Morning Alarm",
  "state": {
    "groups": [
      {
        "id": 101,
        "mute":  false
      }
     ],
    "sources": [
     { 
      "id": 0,
      "input": "stream=1000"
     }
     ]
   }
}
```

You may want to change the group 101 and stream 1000  and source ids here to your specific setup. 

Click the **try** button, the response will have a preset id, you will use that in the node red configuration.

Here's a node red flow that can be imported, it loads the **Morning Alarm** preset every weekday morning at 8am. You will need to change the preset id the the one in the response.

```json
[
    {
        "id": "23d5d4558b1ce636",
        "type": "tab",
        "label": "Flow 1",
        "disabled": false,
        "info": "",
        "env": []
    },
    {
        "id": "e6b9ba40dabda945",
        "type": "http request",
        "z": "23d5d4558b1ce636",
        "name": "",
        "method": "POST",
        "ret": "obj",
        "paytoqs": "ignore",
        "url": "http://amplipi.local/api/presets/10000/load",
        "tls": "",
        "persist": false,
        "proxy": "",
        "insecureHTTPParser": false,
        "authType": "",
        "senderr": false,
        "headers": [],
        "x": 430,
        "y": 160,
        "wires": [
            [
                "14b37f601582ca20"
            ]
        ],
        "info": "Mute All"
    },
    {
        "id": "567b8f7084367c3a",
        "type": "inject",
        "z": "23d5d4558b1ce636",
        "name": "",
        "props": [
            {
                "p": "payload"
            },
            {
                "p": "topic",
                "vt": "str"
            }
        ],
        "repeat": "",
        "crontab": "00 08 * * 1,2,3,4,5",
        "once": true,
        "onceDelay": 0.1,
        "topic": "",
        "payload": "Its time to wake up!",
        "payloadType": "str",
        "x": 140,
        "y": 160,
        "wires": [
            [
                "e6b9ba40dabda945"
            ]
        ]
    },
    {
        "id": "14b37f601582ca20",
        "type": "debug",
        "z": "23d5d4558b1ce636",
        "name": "debug 1",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "payload",
        "targetType": "msg",
        "statusVal": "",
        "statusType": "auto",
        "x": 660,
        "y": 160,
        "wires": []
    }
]
```

After it is loaded it will look something like this:
[image]

You can click the blue button to test it out.

rickballard, 2022-12-02:
When Alexa and Google Assistant are finally supported, scheduled turn-ons
will become simpler.  Google gas a feature called Family Bell in Google
Assistant, to set up smart home events like turning on their smart
speakers, or even just network-connected speakers.  Would this be a better
solution?

Regards, Rick.  416.414.8100

linknum23, 2022-12-02:
There are definitely some things that need to be solidified before we start integrating with Alexa and Google Home. 
We are planning on these but those integrations will take some time...

closed 2022-12-05 by linknum23

rickballard, 2023-01-15:
Thank you for your excellent instructions for adding an alarm clock to AmpliPi, (using node red and AmpliPi's API).  This tutorial is so useful that it should perhaps be made into it's own sticky FAQ DIY/tutorial item . 

You guys could also resell a Home Assistant add-on box, with all your suggested DIY features preloaded...?  Is that idea too redundant, given that AmpliPi can host HA internally?  

Personally, I am not confident abouy altering AmpliPi's internal memory without breaking something, so this is the HA box I ordered as a network add-on:  https://www.crowdsupply.com/nabu-casa/home-assistant-yellow

linknum23, 2023-01-16:
Thanks! I'll see if we can polish this up soon and put it on our discourse.

As for buying a special purpose HA machine, I think that's a great idea. Hosting HA on AmpliPi would probably add significant wear to the compute modules EMMC storage.