micro-nova/AmpliPi #413: Add Power Button for sources

linknum23, 2022-12-08:
This would stop all playback and effectively mute the output of the zones. Both HA and LMS this would be useful as their interfaces each have a built in power button.

linknum23, 2023-01-10:
This is typically mapped to standby functionality, which only happens when all 6 zones are muted now. However we have a fw register that has similar functionality, we can probably map to that.

SteveMicroNova, 2025-04-04:
I've done the `hacs_amplipi` side recently, though my solution doesn't help LMS any.

Zone on/off is handled on the AmpliPi side in #1041 by adding an extra enum for their input that equals negative two, so you can have a disconnected state that is off and another that is still on
Source on/off is handled on the home assistant side by an arbitrary boolean state value that is overridden to on when a stream connects but can be either on or off when nothing is connected
Streams are just like sources, being handled HA side and controlled by an arbitrary boolean that is overridden when connected to a source but can be on despite like of source