micro-nova/AmpliPi #422: Link to 0.1.7 release download is broken

hawkeye312, 2022-12-22:
https://github.com/micro-nova/AmpliPi/releases/download/0.1.7/amplipi-update-0.1.7.tar.gz

Trying to update my firmware which does not yet have the auto-update feature.

linknum23, 2022-12-23:
Sorry about that. We'll fix that soon. There was an issue we had to patch
on all releases and we removed old releases to make it easier.

You should use the 0.1.9 release instead. Here's a link:
https://github.com/micro-nova/AmpliPi/archive/refs/tags/0.1.9-uv-hotfix.tar.gz

Just a heads up, automatic updates are not available yet, however the
updater was improved significantly.

linknum23, 2022-12-23:
Working on this here #423. Looks like main is a little behind on some of our fixes to keep our continuous integration satisfied.

closed 2023-01-10 by linknum23