micro-nova/AmpliPi #427: Announcements disconnect Spotify Connect

kevin1024, 2023-01-01:
While playing a song on Amplipi via Spotify, calling the /api/announce endpoint:

(1) Stops spotify (cool)
(2) Plays my announcement (nice!)
(3) ... that's it.  

I was hoping it would resume playing my song.  I have to reconnect to the Amplipi device in my Spotify client and click play to get it going again.

linknum23, 2023-01-02:
Sorry about that, we are working on fixing issues like this in #262 but it still needs some work to be ready for prime time.

This bug will happen if the announcement uses the same source as the Spotify stream is currently playing. The default source is 4 (id=3) if unspecified.

If the source used for the announcement is not the same, here is the expected sequence that happens when an announcement is played:

1. Spotify is interrupted by the announcement (but keeps playing in the background and to any zones not being announced to)
2. The announcement plays
3. Spotify plays out the zones it was playing before the announcement

Whats happening in your case is:
1. Spotify stream is stopped (and all connections lost)
2. Announcement plays
3. Spotify stream is restarted (but needs to be reconnected to continue playing)

To work around this for now don't use the source used for announcements for anything that needs a persistent remote connection (DLNA, Airplay, Spotify). Instead I would suggest connecting an internet radio stream, Pandora station, or an analog input to that source.

linknum23, 2023-01-10:
I hope this workaround and explanation helps you, until we are able to add better functionality in #262.

closed 2023-01-10 by linknum23

kevin1024, 2023-01-10:
Wow, thank you for the detailed response!  I will give this a try.

linknum23, 2023-01-10:
Oh I figured you already had and just didn't respond. Feel free to reopen this if you have any questions.