micro-nova/AmpliPi #450: Update to 0.2.0 failed, now cannot connect to REST API

tbham, 2023-02-10:
When attempting to update from 0.1.9 to 0.2.0, the update failed. Before trying again, rebooted the RPi, and subsequently, AmpliPi cannot connect to the REST API.

linknum23, 2023-02-10:
Sorry about that! It looks like there may be an issue with your configuration. Can you send us the logs from: http://amplipi.local:19531/entries?

tbham, 2023-02-10:
Here you go. This is a recurring message, noting the restart counter up to 1500+

`Feb 10 07:43:51 amplipi systemd[586]: Started Amplipi Home Audio System.
Feb 10 07:43:54 amplipi authbind[7735]: Traceback (most recent call last):
Feb 10 07:43:54 amplipi authbind[7735]:   File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
Feb 10 07:43:54 amplipi authbind[7735]:     "__main__", mod_spec)
Feb 10 07:43:54 amplipi authbind[7735]:   File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
Feb 10 07:43:54 amplipi authbind[7735]:     exec(code, run_globals)
Feb 10 07:43:54 amplipi authbind[7735]:   File "/home/pi/amplipi-dev/venv/lib/python3.7/site-packages/uvicorn/__main__.py", line 4, in <module>
Feb 10 07:43:54 amplipi authbind[7735]:     uvicorn.main()
Feb 10 07:43:54 amplipi authbind[7735]:   File "/home/pi/amplipi-dev/venv/lib/python3.7/site-packages/click/core.py", line 1137, in __call__
Feb 10 07:43:54 amplipi authbind[7735]:     return self.main(*args, **kwargs)
Feb 10 07:43:54 amplipi authbind[7735]:   File "/home/pi/amplipi-dev/venv/lib/python3.7/site-packages/click/core.py", line 1062, in main
Feb 10 07:43:54 amplipi authbind[7735]:     rv = self.invoke(ctx)
Feb 10 07:43:54 amplipi authbind[7735]:   File "/home/pi/amplipi-dev/venv/lib/python3.7/site-packages/click/core.py", line 1404, in invoke
Feb 10 07:43:54 amplipi authbind[7735]:     return ctx.invoke(self.callback, **ctx.params)
Feb 10 07:43:54 amplipi authbind[7735]:   File "/home/pi/amplipi-dev/venv/lib/python3.7/site-packages/click/core.py", line 763, in invoke
Feb 10 07:43:54 amplipi authbind[7735]:     return __callback(*args, **kwargs)
Feb 10 07:43:54 amplipi authbind[7735]:   File "/home/pi/amplipi-dev/venv/lib/python3.7/site-packages/uvicorn/main.py", line 371, in main
Feb 10 07:43:54 amplipi authbind[7735]:     run(app, **kwargs)
Feb 10 07:43:54 amplipi authbind[7735]:   File "/home/pi/amplipi-dev/venv/lib/python3.7/site-packages/uvicorn/main.py", line 393, in run
Feb 10 07:43:54 amplipi authbind[7735]:     server.run()
Feb 10 07:43:54 amplipi authbind[7735]:   File "/home/pi/amplipi-dev/venv/lib/python3.7/site-packages/uvicorn/server.py", line 50, in run
Feb 10 07:43:54 amplipi authbind[7735]:     loop.run_until_complete(self.serve(sockets=sockets))
Feb 10 07:43:54 amplipi authbind[7735]:   File "/usr/lib/python3.7/asyncio/base_events.py", line 584, in run_until_complete
Feb 10 07:43:54 amplipi authbind[7735]:     return future.result()
Feb 10 07:43:54 amplipi authbind[7735]:   File "/home/pi/amplipi-dev/venv/lib/python3.7/site-packages/uvicorn/server.py", line 57, in serve
Feb 10 07:43:54 amplipi authbind[7735]:     config.load()
Feb 10 07:43:54 amplipi authbind[7735]:   File "/home/pi/amplipi-dev/venv/lib/python3.7/site-packages/uvicorn/config.py", line 318, in load
Feb 10 07:43:54 amplipi authbind[7735]:     self.loaded_app = import_from_string(self.app)
Feb 10 07:43:54 amplipi authbind[7735]:   File "/home/pi/amplipi-dev/venv/lib/python3.7/site-packages/uvicorn/importer.py", line 25, in import_from_string
Feb 10 07:43:54 amplipi authbind[7735]:     raise exc from None
Feb 10 07:43:54 amplipi authbind[7735]:   File "/home/pi/amplipi-dev/venv/lib/python3.7/site-packages/uvicorn/importer.py", line 22, in import_from_string
Feb 10 07:43:54 amplipi authbind[7735]:     module = importlib.import_module(module_str)
Feb 10 07:43:54 amplipi authbind[7735]:   File "/usr/lib/python3.7/importlib/__init__.py", line 127, in import_module
Feb 10 07:43:54 amplipi authbind[7735]:     return _bootstrap._gcd_import(name[level:], package, level)
Feb 10 07:43:54 amplipi authbind[7735]:   File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
Feb 10 07:43:54 amplipi authbind[7735]:   File "<frozen importlib._bootstrap>", line 983, in _find_and_load
Feb 10 07:43:54 amplipi authbind[7735]:   File "<frozen importlib._bootstrap>", line 967, in _find_and_load_unlocked
Feb 10 07:43:54 amplipi authbind[7735]:   File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
Feb 10 07:43:54 amplipi authbind[7735]:   File "<frozen importlib._bootstrap_external>", line 728, in exec_module
Feb 10 07:43:54 amplipi authbind[7735]:   File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
Feb 10 07:43:54 amplipi authbind[7735]:   File "./amplipi/asgi.py", line 27, in <module>
Feb 10 07:43:54 amplipi authbind[7735]:     import amplipi.app
Feb 10 07:43:54 amplipi authbind[7735]:   File "./amplipi/app.py", line 64, in <module>
Feb 10 07:43:54 amplipi authbind[7735]:     from amplipi.ctrl import Api, ApiResponse, ApiCode, RCAs # we don't import ctrl here to avoid naming ambiguity with a ctrl variable
Feb 10 07:43:54 amplipi authbind[7735]:   File "./amplipi/ctrl.py", line 37, in <module>
Feb 10 07:43:54 amplipi authbind[7735]:     import amplipi.streams
Feb 10 07:43:54 amplipi authbind[7735]:   File "./amplipi/streams.py", line 39, in <module>
Feb 10 07:43:54 amplipi authbind[7735]:     from amplipi.mpris import MPRIS
Feb 10 07:43:54 amplipi authbind[7735]:   File "./amplipi/mpris.py", line 10, in <module>
Feb 10 07:43:54 amplipi authbind[7735]:     from dasbus.connection import SessionMessageBus
Feb 10 07:43:54 amplipi authbind[7735]: ModuleNotFoundError: No module named 'dasbus'
Feb 10 07:43:54 amplipi systemd[586]: amplipi.service: Main process exited, code=exited, status=1/FAILURE
Feb 10 07:43:54 amplipi systemd[586]: amplipi.service: Failed with result 'exit-code'.
Feb 10 07:43:55 amplipi systemd[586]: amplipi.service: Service RestartSec=100ms expired, scheduling restart.
Feb 10 07:43:55 amplipi systemd[586]: amplipi.service: Scheduled restart job, restart counter is at 1589.
Feb 10 07:43:55 amplipi systemd[586]: Stopped Amplipi Home Audio System.`

linknum23, 2023-02-10:
Hmm it looks like the pypy package **dasbus** failed to install during the update. This should have caused the update to fail. Either way I don't know why it is not installed. 

For now you should be able to roll back to the previous version 0.1.9 by following the instructions below:

- Go to the updater directly:  http://amplipi.local:5000/update
- Click the Older Releases tab and select the 0.1.9 release
- Click the Start Update button
- Wait for the update to finish. After a successful update it will redirect back to the web app.

tbham, 2023-02-10:
Thanks, managed to rollback to 0.1.9 (via port 5001 - typo above).

Let me know what I can do to repair my system for 0.2.0 and future updates to work.

linknum23, 2023-02-10:
Will do. We haven't been able to reproduce this internally yet.

linknum23, 2023-02-13:
This appears to be due to our python packages not being fixed. We are testing a fix for this now!

If you want to test it out: https://github.com/micro-nova/AmpliPi/releases/tag/0.2.1-alpha.0

linknum23, 2023-02-13:
We have tested this on 3 different AmpliPi's could anyone else test it and see if it fixes this issue/

linknum23, 2023-02-13:
Just in case here's some instructions on how to install the prerelease:

1. Go to the web app at http://amplipi.local/
2. Click the gear icon (gear) in the upper right corner to go to the configuration page
3. Select Updates and click the Check for Updates button
4. Click the Older Releases tab and select the **0.2.1 Testing fixed requirements - alpha** prerelease
5. Click the Start Update button
6. Wait for the update to finish. After a successful update it will redirect back to the web app.

tbham, 2023-02-13:
I'm away from home for the next 10 days. I'll have to wait until then to try the pre-release. Hopefully there are others that can validate in the meantime. Appreciate the response and action. Cheers

linknum23, 2023-02-15:
It looks like this test fixed the issue for one customer.

linknum23, 2023-02-16:
It fixed it for a second user as well. We will make a patch release of this soon. Look out for 0.2.1.

tbham, 2023-02-22:
Successfully updated to 0.2.1 alpha. Thanks again for the assist

linknum23, 2023-03-13:
The source of this issue has been fixed in 0.2.1. Feel free to open this if it happens in that or future releases

closed 2023-03-13 by linknum23