micro-nova/AmpliPi #458: Is there a way to add delay to the stream? If not, there should be.

iloveicedgreentea, 2023-03-28:
Using a setup where I use an AVR to power one zone and amplipi for the rest, the audio is not synced. I need to delay the amplipi by 50-200ms to get it synced up.

linknum23, 2023-04-10:
The underlying airplay stream (shairport-sync) has support for a configurable delay but the rest of the streams supported by AmpliPi do not so we haven't exposed that configuration yet. 

In the future we could expose the delay parameter to the api and the ui. I think that would solve your use case.

In the meantime you have a couple of options:
- modify amplipi's shairport-sync config file generation in /home/pi/amplipi-dev/amplipi/streams.py to add the delay needed
- Route a preamp RCA output from your AVR into one of your AmpliPi's RCA inputs and use that to play to the rest of the zones

iloveicedgreentea, 2023-04-14:
Thanks I will look into that and see if I can help expose that to the UI. Do I need to restart amplipi after changing this?

iloveicedgreentea, 2023-04-14:
I also noticed in the file config code, its using audio_backend_buffer_desired_length under alsa, but in the shairplay docs this is actually listed under general

https://github.com/micro-nova/AmpliPi/blob/main/amplipi/streams.py#L268

https://github.com/mikebrady/shairport-sync/blob/master/scripts/shairport-sync.conf

Seems wrong, unless I am missing something?

iloveicedgreentea, 2023-04-14:
While I think exposing this setting might solve another problem (like lipsync) I don't think it will fix my issue. The airplay stream is going into the amplipi, then one leg is going to the AVR. What I need is a way to delay only the signal in amplipi

If I add _audio_backend_latency_offset_in_seconds_ then it will have the same result, with the AVR signal being delayed.

linknum23, 2023-04-17:
I see that now. I think we misunderstood your setup. 

For this setup the current hardware does not have the capability to correct for delays. This is something we will consider in the future designs.

closed 2023-05-03 by linknum23

iloveicedgreentea, 2023-07-28:
Theres no way to even do it at a system level with pulse or something? I don't care about per stream/source.

linknum23, 2023-07-28:
The current system does not have the capability to do any additional per zone adjustments. We are only able to make audio adjustments per stream.

It might help to look at AmpliPi's system diagram:

[image]

The pi is able to output 4 stereo streaming outputs (source 1-4). Each zone only has control of which source to use and its volume.