micro-nova/AmpliPi #463: Amplipi doesn’t play after failed update

0lek, 2023-04-15:
Hey,

- All was good, then tried update to 0.2.1
- Update failed
- After the update all seems to work, there’s just no sound
- Tried reboot etc, doesn’t help
- Update fails always (also for 0.2.0)
- Current running version is 0.2.0
- logs: https://pastebin.mozilla.org/fpApijbT
- Attachment is the failure msg at update

please advise
[image]

linknum23, 2023-04-17:
Sorry you are having problems with this update.

Looking at your logs your AmpliPi is having trouble downloading the update tarball from GitHub. Here is the relevant failure:

```
Apr 14 12:19:05 amplipi python[771]: downloading update from: https://api.github.com/repos/micro-nova/AmpliPi/tarball/0.2.1
Apr 14 12:19:05 amplipi python[771]: HTTPSConnectionPool(host='api.github.com', port=443): Max retries exceeded with url: /repos/micro-nova/AmpliPi/tarball/0.2.1 (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0x746e24d0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution'))
```

Any chance you can try running the following from an ssh session on your amplipi?

```bash
wget https://api.github.com/repos/micro-nova/AmpliPi/tarball/0.2.1
```

It should succeed. If not there may be a networking issue and you should post the error reported so that we can debug further.

linknum23, 2023-04-17:
As for the sound I don't see anything immediately in your logs.

Can you try doing a full shutdown? The hardware could be in a bad state.

1. go to Settings -> Configuration and Reset
2. Click Shutdown and wait for it to finish
3. Unplug the amplipi
4. plug it back in, waiting 1 minute for startup

linknum23, 2023-04-24:
Did you have any luck on this?

0lek, 2023-04-29:
@linknum23 It is the weirdest thing. It boils down to the Amplipi not being able to resolve DNS...

I updated it to 0.2.1. Had to manually add a DNS server in /etc/resolv.conf 
Then all functions. But I have no clue why it suddenly stopped working, as I have not changed ANYTHING in my network setup, i.e. it used to work in exactly the same local DNS.

Any clues or pointers as to where to look?

linknum23, 2023-05-01:
Wacky. I'll have to add DNS to our troubleshooting steps. Nice work
figuring it out!

closed 2023-05-03 by linknum23