micro-nova/AmpliPi #471: Automate image creation

rtasson, 2023-05-24:
AmpliPi images should be automatically generated, both for build-chain security and robustness of implementation.

rtasson, 2023-07-17:
These images ought to use LVM for snapshots before upgrades & easy reverts.

Lohrer, 2023-08-08:
The [imaging branch](https://github.com/micro-nova/AmpliPi/tree/imaging) contains some changes that might be useful or at least good references.
* [docs/pi_imaging.md](https://github.com/micro-nova/AmpliPi/blob/imaging/docs/pi_imaging.md) was updated with more automated sector count calculations.
* [scripts/bootstrap_pi](https://github.com/micro-nova/AmpliPi/blob/imaging/scripts/bootstrap_pi) adds the Pi user since new Raspberry Pi OS distributions no longer include it.
* [scripts/image_pi](https://github.com/micro-nova/AmpliPi/blob/imaging/scripts/image_pi) I attempted to update this to both read and write images but ran out of time to fully implement reading. I did update the writing to work with compressed images, as well as adding `oflag=dsync` to `dd` which really shouldn't have been left out originally, oops...
* [scripts/image_shrink](https://github.com/micro-nova/AmpliPi/blob/imaging/scripts/image_shrink) This is what I was playing with most recently, and what created the 0.2.1 image.
* [scripts/shrink_image](https://github.com/micro-nova/AmpliPi/blob/imaging/scripts/shrink_image) An offline image shrinking script, now deprecated.

Also, back in the day I did some compression tests to see what we should be using with the 0.1.7 AmpliPi image.
| Algorithm   | Compressed Size (MiB) | Compression Time (s) | Decompression Time (s) |
| -------- | ------ | ------- | ------- |
| xz -9T1  | 1328 |1633.8 |  55.9 |
| xz -9T0  | 1354 | 219.5 |  56.6 |
| 7z      | 1380 | 870.3 |  64.4 |
| xz -6eT0 | 1499 | 191.6 |  64.0 |
| zstd -19T0 | 1624 | 233.6   |  6.4  |
| zstd -9T0 | 1836 | 66.9   |  5.7  |
| bzip2   | 1878 | 281.3 | 120.0 |
| zstd -3 | 1944 | 13.9   |  5.7  |
| gzip     | 2009 | 364.5 |  29.4 |
| Uncompressed | 5592 | 0 | 0 |

rtertiaer, 2024-01-03:
Not explicitly targetting this for 0.4 but 0.5. May do this anyways in the course of producing images for 0.4.