micro-nova/AmpliPi #472: Create a deb repository for compiled dependencies

rtasson, 2023-05-24:
There are a couple issues with binary/system dependency management at the moment (read: not python dependencies.)
* There is a ton of code necessary, in disparate places, to bootstrap and compile dependencies
* The versions of dependencies are not managed
* Compilation is time consuming on a Raspberry Pi, and I've seen it OOM a couple times now (lookin at you, Redsea :eyes: )

To address these concerns, we should cross-compile and package our dependencies that are not available upstream, and host them in our own repository. This should also make it easier to automate image builds, #471 .

Acceptance criteria:
* Packages: shairport-sync, shairport-sync-metadata-reader, redsea, nqptp, gmrender-resurrect
* Platforms: armhf, arm64
* Distributions: buster, bullseye, bookworm
* packages are all signed with a micronova key; perhaps it's reasonable to make a .deb package out of this too, to permit updates.
* package build process is documented
* package hosting is documented

rtertiaer, 2023-12-05:
Most of the work here is done, save for testing. A lot of binaries have been packaged and uploaded in a GCP bucket using reprepro, managed & documented in a currently private git repository (I'll make it public once it's ship-able, but for Those In The Know :tm:  it's at https://github.com/micro-nova/micronova-deb-repo ). We just need to actually install the bucket as a repo on a box and extensively test each built binary. 

I think I'd also like to perhaps distribute this repo as a .deb installable itself, so it can be versioned (ie, if we move away from GCP) and also explicitly dependent on our keyring signature, but that feels like extra credit.

rtertiaer, 2024-01-03:
Not explicitly targeting this for 0.4 but 0.5. May do this anyways in the course of producing images for 0.4.