micro-nova/AmpliPi #475: Spotify Fails to connect

Shwall, 2023-07-09:
Recently, Spotify has stopped connecting to my Amplipi. I can see the device in Spotify after I enable a Spotify stream, but when I try to connect it fails and then disappears from the Spotify available devices. If I reset the stream, it shows up again, but always fails to connect. I tried to factory reset the Amplipi then re-uploaded my config; however, the reset didn't seem to help.  Here is the log for an attempt to connect. Any insights on how to get this working again would be appreciated. 
```
Jul 09 04:39:36 amplipi.local authbind[819]: Loading config from "./config.toml"
Jul 09 04:39:36 amplipi.local authbind[819]: No username specified. Checking username_cmd
Jul 09 04:39:36 amplipi.local authbind[819]: No username_cmd specified
Jul 09 04:39:36 amplipi.local authbind[819]: No password specified. Checking password_cmd
Jul 09 04:39:36 amplipi.local authbind[819]: No password_cmd specified
Jul 09 04:39:36 amplipi.local authbind[819]: No proxy specified
Jul 09 04:39:36 amplipi.local authbind[819]: Using software volume controller.
Jul 09 04:39:36 amplipi.local authbind[819]: no usable credentials found, enabling discovery
Jul 09 04:39:36 amplipi.local authbind[819]: amplipi connected to 1
Jul 09 04:39:36 amplipi.local authbind[819]: Error getting RCA audio status:
Jul 09 04:39:36 amplipi.local authbind[819]:   index out of range
Jul 09 04:39:36 amplipi.local authbind[819]: Error getting RCA audio status:
Jul 09 04:39:36 amplipi.local authbind[819]:   index out of range
Jul 09 04:39:36 amplipi.local authbind[819]: INFO:     192.168.0.54:52540 - "PATCH /api/sources/1 HTTP/1.1" 200 OK
Jul 09 04:39:36 amplipi.local authbind[819]: Error getting RCA audio status:
Jul 09 04:39:36 amplipi.local authbind[819]:   index out of range
Jul 09 04:39:36 amplipi.local authbind[819]: INFO:     192.168.0.54:52540 - "GET /api HTTP/1.1" 200 OK
Jul 09 04:39:37 amplipi.local authbind[819]: Error getting RCA audio status:
Jul 09 04:39:37 amplipi.local authbind[819]:   index out of range
Jul 09 04:39:37 amplipi.local authbind[819]: INFO:     192.168.0.36:39426 - "GET /api HTTP/1.1" 200 OK
Jul 09 04:39:38 amplipi.local authbind[819]: Error getting RCA audio status:
Jul 09 04:39:38 amplipi.local authbind[819]:   index out of range
Jul 09 04:39:38 amplipi.local authbind[819]: INFO:     192.168.0.54:52540 - "GET /api HTTP/1.1" 200 OK
Jul 09 04:39:39 amplipi.local authbind[819]: Error getting RCA audio status:
Jul 09 04:39:39 amplipi.local authbind[819]:   index out of range
Jul 09 04:39:39 amplipi.local authbind[819]: INFO:     192.168.0.36:39426 - "GET /api HTTP/1.1" 200 OK
Jul 09 04:39:39 amplipi.local authbind[819]: Connecting to AP "ap.spotify.com:443"
Jul 09 04:39:40 amplipi.local authbind[819]: Error getting RCA audio status:
Jul 09 04:39:40 amplipi.local authbind[819]:   index out of range
Jul 09 04:39:40 amplipi.local authbind[819]: INFO:     192.168.0.54:52540 - "GET /api HTTP/1.1" 200 OK
Jul 09 04:39:41 amplipi.local authbind[819]: Error getting RCA audio status:
Jul 09 04:39:41 amplipi.local authbind[819]:   index out of range
Jul 09 04:39:41 amplipi.local authbind[819]: INFO:     192.168.0.36:39426 - "GET /api HTTP/1.1" 200 OK
Jul 09 04:39:43 amplipi.local authbind[819]: Error getting RCA audio status:
Jul 09 04:39:43 amplipi.local authbind[819]:   index out of range
Jul 09 04:39:43 amplipi.local authbind[819]: INFO:     192.168.0.54:52540 - "GET /api HTTP/1.1" 200 OK
Jul 09 04:39:43 amplipi.local authbind[819]: Error getting RCA audio status:
Jul 09 04:39:43 amplipi.local authbind[819]:   index out of range
Jul 09 04:39:43 amplipi.local authbind[819]: INFO:     192.168.0.36:39426 - "GET /api HTTP/1.1" 200 OK
Jul 09 04:39:45 amplipi.local authbind[819]: Authenticated as "1261638132" !
Jul 09 04:39:45 amplipi.local authbind[819]: Country: "US"
Jul 09 04:39:45 amplipi.local authbind[819]: Using Alsa sink with format: S16
Jul 09 04:39:45 amplipi.local authbind[819]: Error getting RCA audio status:
Jul 09 04:39:45 amplipi.local authbind[819]:   index out of range
Jul 09 04:39:45 amplipi.local authbind[819]: INFO:     192.168.0.36:39426 - "GET /api HTTP/1.1" 200 OK
Jul 09 04:39:45 amplipi.local authbind[819]: error 403 for uri hm://keymaster/token/authenticated?client_id=2c1ea588dfbc4a989e2426f8385297c3&scope=user-read-playback-state,user-modify-playback-state,user-read-currently-playing
Jul 09 04:39:45 amplipi.local authbind[819]: failed to request a token for the web API
Jul 09 04:39:45 amplipi.local authbind[819]: Error getting RCA audio status:
Jul 09 04:39:45 amplipi.local authbind[819]:   index out of range
Jul 09 04:39:45 amplipi.local authbind[819]: INFO:     192.168.0.54:52540 - "GET /api HTTP/1.1" 200 OK
. . . 
Jul 09 04:41:45 amplipi.local authbind[819]: INFO:     192.168.0.36:39426 - "GET /api HTTP/1.1" 200 OK
Jul 09 04:41:45 amplipi.local authbind[819]: Dbus error getting MPRIS metadata: The name org.mpris.MediaPlayer2.spotifyd.instance2544 was not provided by any .service files
Jul 09 04:41:45 amplipi.local authbind[819]: Dbus error getting MPRIS metadata: The name org.mpris.MediaPlayer2.spotifyd.instance2544 was not provided by any .service files
Jul 09 04:41:45 amplipi.local authbind[819]: Dbus error getting MPRIS metadata: The name org.mpris.MediaPlayer2.spotifyd.instance2544 was not provided by any .service files
Jul 09 04:41:45 amplipi.local authbind[819]: Dbus error getting MPRIS metadata: The name org.mpris.MediaPlayer2.spotifyd.instance2544 was not provided by any .service files
Jul 09 04:41:45 amplipi.local authbind[819]: Dbus error getting MPRIS metadata: The name org.mpris.MediaPlayer2.spotifyd.instance2544 was not provided by any .service filesError getting RCA audio status:
Jul 09 04:41:45 amplipi.local authbind[819]:   index out of range
Jul 09 04:41:45 amplipi.local authbind[819]: INFO:     192.168.0.54:52540 - "GET /api HTTP/1.1" 200 OK
Jul 09 04:41:47 amplipi.local authbind[819]: Error getting RCA audio status:
```

mkomarsr, 2023-08-22:
I've been having similar issues from time to time. I'm running the firmware version with the updated spotify client. I'm happy to gather whatever data might help.

linknum23, 2023-10-04:
The only thing I notice in this log is 
> error 403 for uri hm://keymaster/token/authenticated?client_id=2c1ea588dfbc4a989e2426f8385297c3&scope=user-read-playback-state,user-modify-playback-state,user-read-currently-playing
That looks like an internal error from the Spotify client. @mkomarsr  do you see similar errors when you have issues?

mkomarsr, 2023-10-04:
I'm getting a lot of thisL
Oct 04 16:52:18 amplipi authbind[585]: ALSA lib pcm_direct.c:1824:(_snd_pcm_direct_get_slave_ipc_offset) Invalid value for card
Oct 04 16:52:18 amplipi authbind[585]: playback ch3 open error: No such device
Oct 04 16:52:18 amplipi authbind[585]: Loopback initialization failure.
Oct 04 16:52:18 amplipi authbind[585]: Subprocess exited with non-zero code: 1. Restarting...
Oct 04 16:52:18 amplipi authbind[585]: ALSA lib pcm_direct.c:1824:(_snd_pcm_direct_get_slave_ipc_offset) Invalid value for card
Oct 04 16:52:18 amplipi authbind[585]: playback ch3 open error: No such device
Oct 04 16:52:18 amplipi authbind[585]: Loopback initialization failure.
Oct 04 16:52:18 amplipi authbind[585]: Subprocess exited with non-zero code: 1. Restarting...

Currently, I AM able to connect to it but am not getting audio. Is there by any chance a troubleshooting guide for specific steps to take in diffirent circumstances to help provide better diagnostic data?

Shwall, 2023-10-08:
Thanks for looking into this more, I think I have identified this issue as a combination of faulty mDNS forwarding on my previous Wi-Fi router, and bugs in the Spotify snap installation for Ubuntu 22.04 LTS. Checking for the Spotify mDNS requests on my Ubuntu device connected via Ethernet cable showed up just fine. But the version of Spotify running on that host couldn't connect to the session. After replacing the Wi-Fi router with one that doesn't have mDNS forwarding errors, I could connect to the Amplipi Spotify stream via the android app.  Interestingly, after my Spotify account was connected to the Amplipi stream, the Snap installed Spotify client on my Ubuntu system could identify the device and control it.

closed 2023-10-08 by Shwall