micro-nova/AmpliPi #478: Give the new UI a once-over for fit & finish before marking it 'stable'

rtasson, 2023-07-17:
- [ ] `eslint` all React & implement in Github Actions
- [ ] Create custom build with ElasticAPM/Sentry/etc frontend JS traceback collection and abuse it
- [ ] Look for, file & fix any bugs discovered in the process.

Since 2/3 of the checkboxes here do not have a well-defined stop point, we'll go based on feeling or time elapsed for those.

rtasson, 2023-08-23:
@SteveMicroNova and I had discussed using ElasticAPM RUM in a temporary fashion to hash out bugs. However, after talking this out with Lincoln, I'm pretty sure it makes sense to implement it as an opt-in debug configuration.

Last commit I see [here](https://github.com/micro-nova/AmpliPi/compare/Webapp2-RUM) is [f810437cee](https://github.com/micro-nova/AmpliPi/commit/f810437cee96f482d2996549737b4654235b4749); I'll pick this up from there and run with it.

rtertiaer, 2023-12-11:
I'm closing this particular ticket, with #467 being merged (even though we're still just on release candidates of 0.3.0, it's actual launch.)

closed 2023-12-11 by rtertiaer