micro-nova/AmpliPi #485: More than one announcement at the same time via the api

stamateviorel, 2023-07-27:
The problem is like this:
If an announcement is already playing and another one comes without a volume value there is no sound.
If an announcement is already playing and another one comes with a volume values there is no sound but when the last radio station playing comes back then the volume stays at the value of the second announcement.

This is super annoying because the volume jumps and then stays very loud especially if you were listening at a lower volume.

linknum23, 2023-07-27:
This highlights a race condition in the save/restore handling for the announcement logic. The easiest fix would be to only save the state before the first overlapped announcement and only restore that state after the last overlapped announcement completes.

linknum23, 2023-07-27:
A more advanced announcement interface would allow an overlapping announcement to either queue up (default) or interrupt the current announcement and start playing the new one.

The queue up implementation could use the current logic with a queue in front of it while the interrupting announcement would need to use the "only save first overlapped announcement" logic above.

stamateviorel, 2023-07-28:
well right now i am busy on implementing a queue in openhab so i dont send to may request to the amplipi api. But indeed the amplipi should protect by itself from this. If you need any help let me know

stamateviorel, 2023-07-28:
for the moment i am using the openhab sink function and voicerss to make announcements so this is my workaround the issue from the openhab side for anyone interested 👍 
`const { items, rules, log, triggers } = require('openhab');
const Voice = Java.type("org.openhab.core.model.script.actions.Voice");
const Audio = Java.type("org.openhab.core.model.script.actions.Audio");
const PercentType = Java.type("org.openhab.core.library.types.PercentType");

// Rough estimate of speech speed in words per minute
const wordsPerMinute = 150;

// Store the last processed message
let lastMessage = null;

// Create a custom logger
var logger = log('sayamplipi_logger');

// Function to say the message
//Function to say the message
function sayMessage(message) {
    logger.info(`About to say message: ${message}`);
    try {
        // Play the call.mp3 file before saying the message
        Audio.playSound("call.mp3");

        // Schedule the say command to run after 2 seconds
        if (typeof setTimeout === 'function') {
            setTimeout(() => {
                Voice.say(message, new PercentType(65));
                logger.info(`Message said: ${message}`);
            }, 4000);
        } else {
            logger.error('setTimeout is not a function');
        }
    } catch (error) {
        logger.error(`Error saying message: ${error}`);
    }
    lastMessage = message;
}

// Create a queue to store the commands
let queue = [];

// Function to process the next command in the queue
function processNextCommand() {
    if (queue.length > 0) {
        const [durationMilliseconds, command] = queue.shift();
        if (typeof setTimeout === 'function') {
            setTimeout(() => {
                command();
                processNextCommand();
            }, durationMilliseconds);
        } else {
            logger.error('setTimeout is not a function');
        }
    }
}

// Testing setTimeout
if (typeof setTimeout === 'function') {
    setTimeout(() => logger.info('setTimeout is working'), 1000);
} else {
    logger.error('setTimeout is not a function');
}

// Listen for changes to the 'sayamplipi' item
rules.JSRule({
    name: "Listen to sayamplipi",
    description: "Listens for changes to the 'sayamplipi' item and queues them for processing",
    triggers: [
        triggers.ItemStateChangeTrigger('sayamplipi')
    ],
    execute: (triggerData) => {
        const newValue = items.getItem('sayamplipi').state;
        logger.info(`New value received: ${newValue}`);
        
        // If the new value is null or empty, don't do anything
        if (!newValue) {
            logger.info('sayamplipi changed to null or empty, ignoring');
            return;
        }

        // If the new value is the same as the last processed message, don't do anything
        if (newValue === lastMessage) {
            logger.info('sayamplipi changed to the same value, ignoring');
            return;
        }

        // Calculate the estimated duration of the audio file
        const wordCount = newValue.split(' ').length;
        const durationMinutes = wordCount / wordsPerMinute;
        const bufferTimeMilliseconds = 500; // Half a second
        const durationMilliseconds = durationMinutes * 60 * 1000 + bufferTimeMilliseconds;

        logger.info(`Estimated duration of message: ${durationMilliseconds} ms`);

        // Add the command to the queue
        queue.push([durationMilliseconds, () => sayMessage(newValue)]);

        // If this is the only command in the queue, start processing it
        if (queue.length === 1) {
            processNextCommand();
        }
    }
});
`

linknum23, 2023-07-28:
The main issue in fixing this is that announcements really need to be handled asynchronously to be able to fix this in a reliable way (see #421). This would require a bit of a redesign of the interface which is not on our current plans for the next couple of months.

linknum23, 2025-01-28:
At the moment overlapped announcements can create some leftover streams and presets. Here's an example script to remove them.

#!/bin/bash
dangling_streams=$(curl -s amplipi.local/api | jq '.streams[] | select(.name == "Announcement") | .id')
for d in $dangling_streams; do
  curl -s --request DELETE amplipi.local/api/streams/${d} -o /dev/null && echo "deleted stream ${d}"
done
dangling_presets=$(curl -s amplipi.local/api | jq '.presets[] | select(.name == "PA - announcement") | .id')   
for d in $dangling_presets; do
  curl -s --request DELETE amplipi.local/api/presets/${d}  -o /dev/null && echo "deleted stream ${d}"
done