micro-nova/AmpliPi #489: Create A/B partition flashing partitioning scheme

rtasson, 2023-08-07:
Upgrading AmpliPi software and operating system requirements in-place is sometimes not possible or introduces significant risk. This is the case for switching operating systems, and upgrading from 32bit to 64bit. We should eventually adopt a flashing scheme similar to Android, where we flash a system image to a distinct partition and boot into that, failing back to the prior partition if it does not work. The Raspberry Pi firmware natively supports this functionality through [`tryboot_a_b`](https://www.raspberrypi.com/documentation/computers/config_txt.html#tryboot_a_b). This work supports 

At the moment, we do not need to fully support flashing images to disk from existing software... but the sooner we can change the partition table, the better, as online shrinking of Ext4 is unsupported; the partition scheme, once written is incredibly sticky, while other problems are much more reasonably solved in software further down the line. Considering that there's state to carry over between images, we ought to probably have a small "state" partition at the end of the disk as well.

Currently, AmpliPi system images are ~6gb and take 10-15mins to flash. Upon first boot, `init_resize.sh` will expand first the partition and then the filesystem to fill all empty space on the eMMC. @Lohrer and I discussed the problem space a bit and flshed out a couple of options, each with tradeoffs:
1) Ship two types of images - a single partition containing the update, and a "full" 32GB image that contains all partitions. This enables both Linux and Windows users to un-brick their device (or us to flash/manufacture our own devices) the same way they're doing it now, with the caveat that flashing a full 32GB is extremely time-consuming (perhaps as much as an hour and a half.) However, updates can simply use the smaller single-partition, and we could probably get sophisticated about our flashing too to take less time. This is extremely simply accomplished with our existing `scripts/image_pi`.

2) Ship an image that has two partitions like now, `/boot` and `/`, but supply it with our own custom [`init_resize.sh`](https://github.com/RPi-Distro/raspi-config/blob/408bde537671de6df2d9b91564e67132f98ffa71/usr/lib/raspi-config/init_resize.sh) that creates our extra two partitions upon first boot. This pushes the job of partitioning the devices to the very edge and requires maintaining our own version of `init_resize.sh`; however, it also solves the problem of shipping big images to flash, and also means we only need to ship one image, probably a much better user experience. The updater will probably simply mount the image using `losetup` and shove a new partition where it needs to be. 

Leaning towards option 2, after much hemming & hawing.

linknum23, 2023-08-08:
After talking with @Lohrer 2 makes the most sense to me as well. It is a bigger lift though.

Lohrer, 2023-08-08:
After talking with @linknum23 we'd like to be able to persist user files such as audio files or simple system changes. In order to do that we could create 4 total partitions:
1. /boot
2. /<sup>1</sup>
3. /<sup>2</sup>
4. /home

With this scheme AmpliPi would continue to live in /home  for now, but we'd eventually need to actually package AmpliPi and install to system directories. The two root partitions would allow us to install new system images with the tryboot_a_b system.

closed 2023-09-05 by rtasson