micro-nova/AmpliPi #495: WebApp2 Feedback

SteveMicroNova, 2023-08-16:
Meta issue full of the nitpicks I came across while testing the web2 frontend.

- [ ] Make streams swappable from the player screen
- [ ] Remove dropdown volume bar from the player screen if there's only one group/zone, instead use the volume bar for that group/zone only
- [ ] Make settings tabs more consistent style wise (Ex: every settings page has an X in the top right to go back to the settings page, but the Updates page has [<- Back to App] button in the top left instead, Configuration has the wrong font)
- [ ] Make more things responsive on the Home screen. The X to close a stream glows white when you hover over it, the other clickables should as well
- [ ] Add a "Volume Lock" setting so that individual zones or groups aren't effected by the volume bars that would effect all zones in the group/stream
- [ ] Add a check for if a stream is already being played before it can be added to another source so that you can't accidentally steal music from others (Bonus points if you make the chosen zone tune in to the source already playing the stream)

Lohrer, 2024-01-03:
- Internet radio:
  - Pressing enter on search doesn't search
  - If no stations are found there is no indication that a search was actually performed (at least I think that's what's happening, or maybe it's just not working sometimes? I don't know, there's no status so I don't know what's going on).
  - Search results selection: There is no indication of which station is actually selected after clicking on one of the search results.
  - Inconsistent naming: "logo" should be capitalized to "Logo" to match the other fields. To be consistent "url" would be "Url" but that's weird so it should probably be "URL".
  - Nice-to-have: after entering a logo on the Internet Radio stream creation modal, it would be awesome to see that logo as a thumbnail next to the "logo" text.
  - Nice-to-have: All the text fields including the two URL sections are fixed-width. It would be very helpful to have them expand to the width of the modal.
- There is no playing/paused/stopped status on the "Home" view, so it's impossible to see which streams are actually playing without 3 clicks per enabled stream.
- Should we replace the text under "HW reboot" from "Raspberry Pi-based controller" to Embedded Linux controller or something? Or are we not caring about removing RPi from branding?
- There's a dot on the settings icon when an update is available, but there's nothing to inform the user that that's what the dot actually means. Maybe add a dot next to updates after selecting the settings page?

linknum23, 2024-01-05:
I added all of those as issues. Thanks for the feedback @Lohrer

SteveMicroNova, 2024-01-05:
With many of these UI issues we ought to bias towards using [MaterialUI](https://mui.com/material-ui/) since that's a package we already use and has premade components that can do what we need, such as the [snackbar component](https://mui.com/material-ui/react-snackbar/)'s ability to function as a notification/warning popup, something that covers at least two of the issues on this list

rtertiaer, 2024-05-15:
I've assigned a label of usually `webapp` and sometimes `user-experience` to this. This ticket should not be maintained moving forward. Please use tags.

closed 2024-05-15 by rtertiaer