micro-nova/AmpliPi #498: Make settings tabs more consistent style wise (Ex: every settings page has an X in the top right to go back to the settings page, but the Updates page has [<- Back to App] button in the top left instead, Configuration has the wrong font)

SteveMicroNova, 2023-08-16:


SteveMicroNova, 2024-06-28:
Since writing this issue, I've been made aware that the Updates page may be separate and different on purpose, I'd like to affirm whether or not we want to actually change it @linknum23

Changing it would require either merging it into our current react app so it can make use of the same stylings, or making it into its own react app and copying a lot of the same style info over

linknum23, 2024-06-28:
The updater is a completely separate app. Migrating it would be a fair amount of work.
 I like the idea of making it look pretty like our current UI, but I think we may want to wait on our changes to the updater for imaging to avoid doing extra work.

SteveMicroNova, 2024-06-28:
> I think we may want to wait on our changes to the updater for imaging to avoid doing extra work.

Got it, splitting this issue out and adding the Updater page portion to #780

closed 2024-08-08 by SteveMicroNova