micro-nova/AmpliPi #523: No output.

bangell, 2023-10-17:
I have a unit from when I first went out on kickstarter.  I have used it and it has been great, however past month i haven't gotten anything output.  I have tried the 2.1 firmware even.  still nothing.  the unit is visible on the network.  I can talk to it, i can connect through airport and bluetooth, just nothing.  I tried the groove station, nothing.  It's like the amplifier section is just off.

Anyway I can verify?

linknum23, 2023-10-17:
Oh no sorry to hear that. We'll get this sorted out.

First thing lets try is a full shutdown/reboot of the system:
1. Click Settings (⚙) -> Configuration and Reset -> Hardware Shutdown (Power down device) -> Shutdown
2. Unplug power to the unit and wait several seconds
3. Plug power back in

That may fix the problem, however if it still exists please send us a system log so we can trouble shoot this further.
Use the following link to get to the logs: http://amplipi.local:19531/entries , they are also accessible from logs link at the bottom of the AmpliPi ui.

Our email is support@micro-nova.com, you can also post your logs to this issue if it makes more sense.

bangell, 2023-10-17:
> Looking at your logs the main issue we are seeing is that the USB audio device CH3 (source 4) is not being detected. If this persists across a full shutdown/reboot there may be a hardware issue causing the USB audio outputs to not start up.
>
> Lets diagnose this further, can you do the following and report back?
>
> Check if there is any devices plugged into your AmpliPi's USB ports. Try removing them and rebooting the unit. Does audio output work?
> Try playing audio out Source 1. Since it is driven by a separate non-USB device, it may work. Does that work?
> For others experiencing similar issues: here is the relevant part of the log:
>
> Oct 17 15:29:17 amplipi authbind[846]: [02453c70] alsa audio output error: cannot open ALSA device "ch3": No such device
> ```

ok some update:

i had a bluetooth usb installed.  I removed it and still nothing.
So then I tried moving something into RCA input 1, to see if it works.

sure enoug that worked, and now everything works????  I am not sure why though

Lohrer, 2023-10-18:
I don't think you've fixed your original problem. If you're selecting RCA1/Input 1 (sorry I forget what the GUI shows for the exact name, don't have an AmpliPi handy) as the audio source then the audio never makes its way onto the Control board with the Pi and audio DACs. It simply is routed as analog audio to the amp zones you selected. You can see how the audio flows through the system with [this diagram](https://github.com/micro-nova/AmpliPi/blob/main/hw/diagrams/signal_diagram.drawio.svg) if you're interested.

The issue you mentioned originally, Airport, Bluetooth, and Groove Salad, are all digital sources that have to go through a DAC before they make it to our analog audio mux. The error Lincoln spotted is saying one of those DAC channels isn't being found by the Pi.

So far, it seems like the amps work, the Pi and Ethernet work, and the analog audio mux works. On the other hand, we know at least one audio DAC channel isn't working and as Lincoln mentioned the last 3 channels are all on one USB DAC. So most likely all 3 of those don't work. It would still be useful to know if the first DAC channel works.
>* Try playing audio out Source 1. Since it is driven by a separate non-USB device, it may work. Does that work?

bangell, 2023-10-20:
Lohrer, after you stated that, i went back and you are correct the 3 other channels still do not produce output.  So I believe you are right about the issue I am seeing.  Line 1 works, line 2, 3, 4 do not.