micro-nova/AmpliPi #525: Feature: Add File Browser

SteveMicroNova, 2023-10-26:
Related to #66, somewhat

Add a local file browser and player so that a user can play their own music, either from a network drive or an external drive connected via USB

Make a good UI on the frontend for this that can also be used for TiDAL, Pandora Premium, and all other stream types that require the user to select what is being played rather than simply tuning in

rtertiaer, 2024-06-25:
#692 implemented a methodology for this; we mostly just need a backend for this.

 I hesitate a bit on this one, only because the local filesystem should be considered somewhat privileged.

SteveMicroNova, 2024-06-26:
> I hesitate a bit on this one, only because the local filesystem should be considered somewhat privileged.

I think the intended solution was to allow use of NAS storage or USB drives, not actually touching the local drive at all
I feel if we make a "FileBrowser" user that just has access to mounted drives (potentially by modifying my udisks2-listener.service to work outside of LMS mode) it should be very quick, easy, and secure

SteveMicroNova, 2024-09-26:
This has been completed by #764
We still need to add NAS support, but that is a different issue.

closed 2024-09-26 by SteveMicroNova