micro-nova/AmpliPi #538: Api difference between 0.2.1 and 0.3.0 beta

suttin, 2023-11-22:
Hello, 

The webserver that is serving /api appears to auto redirect in 0.2.1 if you navigate to /api/

In 0.3.0 beta the path does not auto redirect. I don't know what the intended behavior is, but it did break this Home-assistant integration, https://github.com/brianhealey/hacs_amplipi.

linknum23, 2023-11-22:
This behaviour should be fixed in the latest 0.3.0 beta released a couple
of weeks ago, 0.3.0-beta.3.

You can check your version via settings->About (Note that some earlier beta
releases reported as 0.2.1.*RUM*).

Please try it out and let us know if there are still issues.

Lohrer, 2023-12-11:
I just checked this again in [0.3.0-rc.0](https://github.com/micro-nova/AmpliPi/releases/tag/0.3.0-rc.0) and both `/api` and `/api/` seem to work. Is there any issue here still?

closed 2023-12-11 by linknum23