micro-nova/AmpliPi #547: Create & ship a localhost script that unsets the admin password

rtertiaer, 2023-12-04:
You should be able to remove or reset the admin password from the command line.

SteveMicroNova, 2024-10-15:
This might also be good to add to the Admin Panel added in #828, have the script be accessible via an endpoint and a subprocess wrapper as well

rtertiaer, 2024-11-04:
>This might also be good to add to the Admin Panel added in https://github.com/micro-nova/AmpliPi/pull/828, have the script be accessible via an endpoint and a subprocess wrapper as well

how is this any different from the existing admin password un/setting in the admin panel?

SteveMicroNova, 2024-11-04:
I don't view that password change thing as an "admin password", it is only used for the frontend and is not used for any of the internals; it is not used to ssh, it is not used for sudo, as far as I know all it does is simply lock down the frontend until you enter the password.

rtertiaer, 2024-11-06:
ah, sorry, I didn't make this ticket terribly clear; I'll edit. This ticket is to add a script to reset the frontend password from the backend. The way you reset a UNIX user password on the command line is a command called `passwd` - no need for a 'script' :)