micro-nova/AmpliPi #572: Disallow saving Pandora stream configs without all fields completed

rtertiaer, 2023-12-19:
The UI will gleefully send off your request with no "name" field; the API rightfully rejects it. we can both handle the form better before it hits the API, and also handle the 422 Unprocessable Request response from the API.

rtertiaer, 2024-01-02:
It seems a great deal of error handling is missing here; in part, this is simply because there are no facilities built in to this app to actually handle and display errors to users. Currently fussing with the best way to address that with this issue.

rtertiaer, 2024-01-08:
also add a description field

SteveMicroNova, 2024-01-08:
Related to our meeting from this morning, I think this issue is the one we were talking about this for. While you're already screwing around with the form this would be added to, here's some added directions:
Add explanation for how to find a pandora station ID, even if you just use a [mui tooltip](https://mui.com/material-ui/react-tooltip/) that contains a short how-to with the [image](https://github.com/micro-nova/AmpliPi/blob/main/docs/imgs/app_demos/pandora_url_id.png) from the recently updated [webapp guide](https://github.com/micro-nova/AmpliPi/blob/main/docs/WEB_APP.md)

closed 2024-03-04 by rtertiaer