micro-nova/AmpliPi #580: No confirmation dialog for options on config page

gorski123, 2023-12-21:
When uploading a new config, resetting the box back to factory config, performing a HW reset, etc, It would probably be a good idea to first show a pop-up dialog confirming the choice. Then there should be some type of spinner while the action is performed, followed by some type of confirmation that the selected action has been completed.

SteveMicroNova, 2023-12-21:
This should be fairly easy to achieve, we already use the React library that contains a [loading spinner](https://mui.com/material-ui/react-progress/#circular)

I'm not able to spend time to this rn but there's the solution for whenever someone takes on this issue

rtertiaer, 2024-05-15:
probably deserve to do the same with LMS mode, particularly beause it downloads some config

PR #763 (2024-06-27): ResponsiveConfigPage
### What does this change intend to accomplish?
Closes #580 by adding confirmation modals, loading spinners, success notifications

Also atomized/componentized the individual parts of the config page while I was at it, now it will be easier to add more sections (or edit them all at once) in the future

### Checklist

* [x] Have you tested your changes and ensured they work?
* [x] Have you checked to ensure there aren't other open [Pull Requests](../../../pulls) for the same update/change?
* [x] If applicable, have you updated the CHANGELOG?
* [x] Does your submission pass  [... cut]
closed 2024-06-27 by SteveMicroNova