micro-nova/AmpliPi #586: Validate the new UI functions under Safari

rtertiaer, 2023-12-22:
Micro-Nova engineers seldomly use Safari; in using it today, it appeared to be non-functional. We should follow-up and validate if that wasn't a strange edge-case, or if this doesn't function in Safari.

SteveMicroNova, 2023-12-22:
I'd like to ask for some clarification on this one, if it's the desktop or mobile version of Safari, and how "non-functional" is quantified (outright doesn't load, missing some features, etc.)
I have an iPhone so I naturally also use the mobile version of safari and haven't noticed any glaring issues in my experience

rtertiaer, 2023-12-22:
I use mobile safari often. This was a desktop version of Safari. The page was simply black, in both a regular and private browsing window (to try to get it to not use caching, but I don't knwo safari.)

rtertiaer, 2024-01-03:
Safari 13.1.2 in High Sierra breaks, stating:
> SyntaxError: Unexpected identifier 'fetch'. Expected ')' to end an argument list.

I'm noting that this version doesn't support [top level await](https://caniuse.com/?search=top%20level%20await), which [may be relevant here](https://github.com/micro-nova/AmpliPi/blob/f1a39a6cd4676e59f2eef4f870a43e2f5844e6b9/web/src/App.jsx#L20).

rtertiaer, 2024-01-03:
I've tested with removing that chunk of code, and yeah - it works great in old Safari. I'll probably just rip out the await and go about it with a `.then()`.

rtertiaer, 2024-01-08:
Gate debug stuff behind a debug flag

rtertiaer, 2024-01-09:
* cannot use `npm build --omit=dev` because that omits `vite`, which is necessary to build
* cannot use `npm build --omit=optional`, because esbuild uses optional dependencies to download the correct binary per host 🥴 
I'm tempted to just remove elasticapm full stop.

linknum23, 2024-01-09:
In that case it probably makes sense to design it in when/if we need it
again.

closed 2024-01-17 by rtertiaer