micro-nova/AmpliPi #591: Webapp build instructions

Lohrer, 2023-12-28:
I tried to run the deploy script but it fails with `./scripts/deploy: line 150: npm: command not found`. Currently, for other dependencies the deploy script does it's best to auto-install, but at the very least some info about what's missing should be printed. For example, which npm version to install? On Debian/Ubuntu install with an apt package? Pip? Something else?

I did check doc/DEVELOPING.md and web/readme.md for any info on npm or building the webapp, but couldn't find anything.

I then tried to build on an AmpliPi with the steps in scripts/deploy, figuring the npm installed on AmpliPi (5.6.0) would work:
```sh
cd amplipi-dev/web
npm install
# npm run build
```

npm install failed when trying to install `esbuild@0.18.20`.

So I would like to request some documentation on how to build the webapp and/or run `scripts/deploy` to install a development AmpliPi version.

linknum23, 2023-12-28:
I have fought with node in the past and ended up using this thing called nvm to install a specific version of node: https://github.com/nvm-sh/nvm

It looks like my system uses node **v14.15.0** to build the webapp. So node 14 is probably sufficient.

So installing node with nvm looks something like this:
```
# install the latest? nvm to manage versions of node
curl -o- https://raw.githubusercontent.com/nvm-sh/nvm/v0.39.7/install.sh | bash
# install and use node 14
nvm use 14
```

Installing this added silly stuff to my .bashrc. I made sure to add a comment indicating its ridiculousness so I wouldn't delete it.
```
# js is silly
 export NVM_DIR="$HOME/.nvm"
 [ -s "$NVM_DIR/nvm.sh" ] && \. "$NVM_DIR/nvm.sh"  # This loads nvm
 [ -s "$NVM_DIR/bash_completion" ] && \. "$NVM_DIR/bash_completion"  # This loads nvm bash_completion
```

linknum23, 2023-12-28:
There may be a better way to do this.

rtertiaer, 2024-01-02:
Good catch! this appears to be something missed from #533 . 

I'll add this to the 0.3 milestone; should be a simple fix to at least check and document this better.

Lohrer, 2024-01-02:
This is being added to DEVELOPING.md but to echo it here as well, turns out we need at least Node.js >=18. Anything older is deprecated. If the version of `nodejs` in apt is outdated, using NVM as Lincoln suggests is an easy alternative. Not trusting 'random script from the internet', on Ubuntu 20.04 I followed NVM's readme and simply ran:

```sh
NVM_DIR="${HOME}/.nvm"

git clone --branch v0.39.7 https://github.com/nvm-sh/nvm.git "$NVM_DIR"
cat >> "${HOME}/.bashrc" << EOF

[ -s "$NVM_DIR/nvm.sh" ] && source "$NVM_DIR/nvm.sh"  # This loads nvm
[ -s "$NVM_DIR/bash_completion" ] && source "$NVM_DIR/bash_completion"  # This loads nvm bash_completion
EOF
source "$NVM_DIR/nvm.sh"
nvm install 18
```

closed 2024-01-03 by rtertiaer