micro-nova/AmpliPi #599: Update updater to flash full images

rtertiaer, 2024-01-03:


rtertiaer, 2024-07-22:
what i envisioned, at a high level:

-     the amplipi requests a list of available updates for it from an upstream api, sending in its current hw/fw/sw revs
-     the upstream api returns a list of versions available for that wombo-combo and where to grab a more detailed manifest for a particular version
-     the amplipi grabs that manifest. it contains checksums, gpg signatures for the image itself, but importantly it also has a list of pre- & post-install commands
-     during upgrade, pre-commands are run "first" (perhaps concurrently to flashing?), then dding the image, then either chroot & run post-install & OR reboot & run post-install.
-     maybe send an 'upgrade successful' response back to the API, so we can spot things before they become issues

SteveMicroNova, 2026-09-18:
This is almost entirely done on #1106, minus the e-ink portion.