micro-nova/AmpliPi #614: Re-add Pandora controls

SteveMicroNova, 2024-01-08:
During the creation of the new webapp, we forgot to readd the buttons that let you like/dislike a specific song. The backend api endpoints still exist for this, it's just a matter of adding the buttons to pandora streams specifically

SteveMicroNova, 2024-03-11:
Currently looking into this, it seems we're in need of re-upping our API documentation as I've just seen that there are "love" and "ban" commands that are Pandora-exclusive and many a JSON saying as much but I couldn't find an instance of those being used

Perhaps my cursory glance was too fast, my ctrl-f-ing too hasty; but I've had many reports of the API docs being less than stellar lately

rtertiaer, 2024-04-19:
#667 closed this

closed 2024-04-19 by rtertiaer