micro-nova/AmpliPi #625: Sources doubling up audio

SteveMicroNova, 2024-01-19:
Related to [this post](https://amplipi.discourse.group/t/streams-doubling-up-when-first-starting/283) in our discourse group

We've been hearing reports of various stream types "doubling up" (seemingly starting twice when loaded) since the 0.3.0 release
If anyone else has this issue, please collect logs via Settings->About->Logs and send them to us at [support@micro-nova.com](mailto:support@micro-nova.com) with the subject "RE: stream echo"

As for the actual issue documentation, streams are loading twice. This used to be an issue with just internet radio (#400) but it applies to multiple now, with reports of LMS and I personally have heard it happen with pandora

This _feels_ like it's related to frontend react state management, but I can't promise that since I haven't poked around yet

rtertiaer, 2024-01-22:
We spotted this once in-house; after poking at it, I don't believe it to be explicitly related to frontend state management. Couple of notes:
* This happened immediately after an upgrade to 0.3.2. This was reported before 0.3.2, so it wasn't introduced with it. Part of the upgrade is a reboot, effectively re-instantiating the stream.
* When this happens, it was happening to only one internet radio stream from VLC; the one that the UI choses for source 4/index 3. (Perhaps #626 could alleviate this from occurring, if this has to do with the other DAC.)
* There was only one instance of VLC running to that source. 
* Removing & restarting the stream did indeed fix it, even to the same source.
* A simple reboot did not trigger this. Reprogramming the firmware and rebooting did not trigger this. Reinstalling 0.3.2 did not trigger this.

rtertiaer, 2024-01-22:
This seems to relatively reliably happen with an internet radio station configured for http://stream.laut.fm/eurobeat . I can't repro on my localhost VLC.

kjkoning, 2024-02-04:
I've experienced this issue as well and was able to fix it by commenting out lines 208 and 225 of runvlc.py (restart_vlc()). I think only one of these lines is the actual cause, but I did both to be safe. It looks like somewhere during startup of VLC, it errors and restarts VLC, but it doesn't stop the already running VLC instance, so it ends up with two running within the same runvlc.py process. A better fix would likely be to stop VLC in the restart_vlc() function prior to restarting it.

rtertiaer, 2024-02-28:
This was a very helpful hint @kjkoning ! One of the reasons `restart_vlc()` is called has logs that look like this:
```
Error: name 'latest_info' is not defined
Waiting to restart vlc
Attempting to restart VLC
```

It appears that `latest_info` is used out of scope  [here](https://github.com/micro-nova/AmpliPi/blob/313cad429c7ae75c1b971c9be04cc5b586d5b6fd/streams/runvlc.py#L198); I'll have a fix coming for that shortly.

There's another note here - this feels like something our linting should have picked up, but a) we don't lint these things, and b) it doesn't :thinking: I'm going to point `pylint -E` at these anyways, because that feels like a blind spot otherwise.

rtertiaer, 2024-03-04:
for folks following along at home: this will get pushed out in a point release Very Soon :tm: , possibly tomorrow.

closed 2024-03-04 by rtertiaer

rtertiaer, 2024-03-06:
Released in 0.3.4!