micro-nova/AmpliPi #636: Pandora Stream issues in 0.3.x

datagitlies, 2024-02-05:
After updating from `0.2.x` to `0.3.x` I've been having some problems consistently playing any Pandora streams.  Restarting the unit is the only way I've been able to recover thus far.  I am currently running version `0.3.3` 

So here's the scenario:

1. Open the AmpliPi App (Android).
2. Click the ➕ 
3. Choose a Pandora Stream and a Zone
4. Click the ✔️ 
5. App looks normal but nothing happens, no audio for that Zone (but the unit shows blue light for the Zone)
6. Wait a while, songs aren't changing according to the "Player" section of the App
7. Via the "Player" click the next button ... again, nothing happens (no audio etc).

Context - In the recent past (like the day prior) I had the unit playing the selected Pandora Stream on the Selected zone.

I decided to ssh into the unit and peek at some logs via `journalctl`.  Only one interesting find was that `next` click resulted in a 404 due to the error `Failed to execute stream command: next: Command next failed to send: [Errno 2] No such file or directory: ''`.  If there's somewhere else in the logs I can / should be looking I'd be happy to share those logs here.

```
Feb 05 11:36:43 amplipi authbind[857]: INFO:     192.168.2.133:45324 - "GET /api/sources HTTP/1.1" 200 OK
Feb 05 11:36:48 amplipi authbind[857]: INFO:     192.168.2.133:45326 - "GET / HTTP/1.1" 304 Not Modified
Feb 05 11:36:48 amplipi authbind[857]: INFO:     192.168.2.133:45326 - "GET /assets/index-03867d4c.js HTTP/1.1" 304 Not Modified
Feb 05 11:36:48 amplipi authbind[857]: INFO:     192.168.2.133:45328 - "GET /assets/index-a8833185.css HTTP/1.1" 304 Not Modified
Feb 05 11:36:49 amplipi authbind[857]: INFO:     192.168.2.133:45328 - "GET /assets/amplipi_logo-936651c4.png HTTP/1.1" 304 Not Modified
Feb 05 11:36:49 amplipi authbind[857]: INFO:     192.168.2.133:45330 - "GET /assets/favicon-d69299aa.png HTTP/1.1" 304 Not Modified
Feb 05 11:36:49 amplipi authbind[857]: INFO:     192.168.2.133:45326 - "GET /api HTTP/1.1" 200 OK
Feb 05 11:36:49 amplipi authbind[857]: INFO:     192.168.2.133:45326 - "GET /assets/opensans_v35-325f509b.ttf HTTP/1.1" 304 Not Modified
Feb 05 11:36:50 amplipi authbind[857]: INFO:     192.168.2.133:45326 - "GET /api HTTP/1.1" 200 OK
Feb 05 11:36:50 amplipi authbind[857]: INFO:     192.168.2.133:45326 - "PATCH /api/zones HTTP/1.1" 200 OK
Feb 05 11:36:50 amplipi authbind[857]: INFO:     192.168.2.133:45330 - "GET /assets/rca_inputs-6123dfcb.jpg HTTP/1.1" 304 Not Modified
Feb 05 11:36:50 amplipi authbind[857]: INFO:     192.168.2.133:45328 - "GET /assets/internet_radio-42b8fc4a.png HTTP/1.1" 304 Not Modified
Feb 05 11:36:50 amplipi authbind[857]: INFO:     192.168.2.133:45334 - "GET /assets/raleway_v28-ddb04dad.ttf HTTP/1.1" 304 Not Modified
Feb 05 11:36:50 amplipi authbind[857]: INFO:     192.168.2.133:45332 - "GET /assets/pandora-5f2fd489.png HTTP/1.1" 304 Not Modified
Feb 05 11:36:52 amplipi authbind[857]: INFO:     192.168.2.133:45332 - "GET /api HTTP/1.1" 200 OK
Feb 05 11:36:53 amplipi authbind[857]: INFO:     192.168.2.133:45332 - "GET /api HTTP/1.1" 200 OK
Feb 05 11:36:55 amplipi authbind[857]: INFO:     192.168.2.133:45332 - "GET /api HTTP/1.1" 200 OK
Feb 05 11:36:55 amplipi authbind[857]:   starting connection via: /home/pi/amplipi-dev/venv/bin/python /home/pi/amplipi-dev/streams/process_monitor.py alsaloop -C lb0p -P ch3 -t 100000
Feb 05 11:36:55 amplipi authbind[857]: INFO:     192.168.2.133:45332 - "PATCH /api/sources/3 HTTP/1.1" 200 OK
Feb 05 11:36:55 amplipi authbind[857]: INFO:     192.168.2.133:45332 - "PATCH /api/zones HTTP/1.1" 200 OK
Feb 05 11:36:55 amplipi authbind[857]: INFO:     192.168.2.133:45334 - "PATCH /api/zones HTTP/1.1" 200 OK
Feb 05 11:36:56 amplipi authbind[857]: INFO:     192.168.2.133:45334 - "GET /api HTTP/1.1" 200 OK
Feb 05 11:36:58 amplipi authbind[857]: INFO:     192.168.2.133:45334 - "GET /api HTTP/1.1" 200 OK
Feb 05 11:36:59 amplipi authbind[857]: INFO:     192.168.2.133:45334 - "GET /api HTTP/1.1" 200 OK
Feb 05 11:37:01 amplipi authbind[857]: INFO:     192.168.2.133:45334 - "GET /api HTTP/1.1" 200 OK
Feb 05 11:37:02 amplipi authbind[857]: INFO:     192.168.2.133:45334 - "GET /api HTTP/1.1" 200 OK
Feb 05 11:37:04 amplipi authbind[857]: INFO:     192.168.2.133:45334 - "GET /api HTTP/1.1" 200 OK
Feb 05 11:37:05 amplipi authbind[857]: INFO:     192.168.2.133:45334 - "GET /api HTTP/1.1" 200 OK
Feb 05 11:37:07 amplipi authbind[857]: INFO:     192.168.2.133:45334 - "GET /api HTTP/1.1" 200 OK
Feb 05 11:37:08 amplipi authbind[857]: INFO:     192.168.2.133:45334 - "GET /api HTTP/1.1" 200 OK
Feb 05 11:37:10 amplipi authbind[857]: INFO:     192.168.2.133:45334 - "GET /api HTTP/1.1" 200 OK
Feb 05 11:37:11 amplipi authbind[857]: INFO:     192.168.2.133:45334 - "GET /api HTTP/1.1" 200 OK
Feb 05 11:37:13 amplipi authbind[857]: INFO:     192.168.2.133:45334 - "GET /api HTTP/1.1" 200 OK
Feb 05 11:37:14 amplipi authbind[857]: INFO:     192.168.2.133:45334 - "GET /api HTTP/1.1" 200 OK
Feb 05 11:37:16 amplipi authbind[857]: INFO:     192.168.2.133:45334 - "GET /api HTTP/1.1" 200 OK
Feb 05 11:37:16 amplipi authbind[857]: INFO:     192.168.2.133:45334 - "PATCH /api/zones/5 HTTP/1.1" 200 OK
Feb 05 11:37:16 amplipi authbind[857]: INFO:     192.168.2.133:45336 - "PATCH /api/zones/5 HTTP/1.1" 200 OK
Feb 05 11:37:16 amplipi authbind[857]: INFO:     192.168.2.133:45338 - "PATCH /api/zones/5 HTTP/1.1" 200 OK
Feb 05 11:37:17 amplipi authbind[857]: INFO:     192.168.2.133:45338 - "GET /api HTTP/1.1" 200 OK
Feb 05 11:37:19 amplipi authbind[857]: INFO:     192.168.2.133:45338 - "GET /api HTTP/1.1" 200 OK
Feb 05 11:37:20 amplipi authbind[857]: INFO:     192.168.2.133:45338 - "GET /api HTTP/1.1" 200 OK
Feb 05 11:37:22 amplipi authbind[857]: INFO:     192.168.2.133:45338 - "GET /api HTTP/1.1" 200 OK
Feb 05 11:37:23 amplipi authbind[857]: INFO:     192.168.2.133:45338 - "GET /api HTTP/1.1" 200 OK
Feb 05 11:37:23 amplipi authbind[857]: Error: Failed to execute stream command: next: Command next failed to send: [Errno 2] No such file or directory: ''
Feb 05 11:37:23 amplipi authbind[857]: INFO:     192.168.2.133:45338 - "POST /api/streams/1002/next HTTP/1.1" 404 Not Found
Feb 05 11:37:25 amplipi authbind[857]: INFO:     192.168.2.133:45338 - "GET /api HTTP/1.1" 200 OK
Feb 05 11:37:26 amplipi authbind[857]: INFO:     192.168.2.133:45338 - "GET /api HTTP/1.1" 200 OK

```

rtertiaer, 2024-02-05:
This is very possibly related to & fixed by #570 . It sounds the directory we create for Pandora logging/controls/etc doesn't exist anymore because Pandora is crashing or failing.

aron7676, 2024-03-11:
I too am having this issue, I have updated to latest version available. Anything I can pull from logs?

datagitlies, 2024-05-25:
@rtertiaer I just installed `0.3.5` and will report back if I see an improvement.

SteveMicroNova, 2025-04-04:
> [@rtertiaer](https://github.com/rtertiaer) I just installed `0.3.5` and will report back if I see an improvement.

Hey @datagitlies!

I'm going through our github issues list to do some spring cleaning, can you report back if you still have this issue? I'd hate to assume and close it if you've still got trouble.

datagitlies, 2025-04-04:
> Hey [@datagitlies](https://github.com/datagitlies)!
>
> I'm going through our github issues list to do some spring cleaning, can you report back if you still have this issue? I'd hate to assume and [...]

Yes, updating solved my issue.  I'm currently on `0.4.4` and I am no longer having any issues like the above.  I'll be updating to `0.4.6` soon!

closed 2025-04-04 by datagitlies