micro-nova/AmpliPi #638: Create automated full-unit tests

Lohrer, 2024-02-07:
We would like to improve testing speed and accuracy by automating as many of the full-unit tests as possible. To start just the AmpliPi main unit should be targeted, but eventually the expanders and streamer should have automated tests as well. The end result should be as many of the checkboxes in [TESTING.md](https://github.com/micro-nova/AmpliPi/blob/main/hw/TESTING.md) being automatically checked (or not) with minimal human intervention.

Tests that should be automated:
- Configure EEPROM (Needs values input by user)
- USB Ports
- Ethernet
- Program Main + Exp Preamp
- Communicate with expander
- Aux Input
- Preamp
- Peak detect
- Preouts

Tests that will be difficult to automate:
- Service Port
- Display
- LEDs
- Fans and Power

- [ ] Get hardware capable of performing the above tests
  - [ ] Find ADCs to record the 32 audio channel outputs
  - [ ] Find DACs to output to the 10 audio channel inputs
  - [ ] Find/build speaker-level to line-level converter for 12 channels.
  - [ ] Computer system capable of processing the audio
- [ ] Write software to record a sample of audio for all possible configurations.
- [ ] Write software to analyze the recorded samples.

Lohrer, 2024-02-07:
This could be a good ADC/DAC https://motu.com/en-us/products/gen5/ultralite-mk5/ but it's only got 6 input channels out of the 32 we need. Maybe we could do some analog muxing?

There's also this beast with 16 analog inputs https://us.focusrite.com/products/red-16line. Or this cheaper one with 8 analog inputs but it doesn't look like they are fixed https://us.focusrite.com/products/clarett-plus-8pre.

linknum23, 2024-02-08:
> @linknum23 <https://github.com/linknum23> suggested looking at the
> Pi+HiFiBerry DAC/ADC. Here's some options:
>
> DAC+ADC options:
>
>    - $50 HiFiBerry DAC+ ADC
>    <https://www.hifiberry.com/shop/boards/hifiberry-dac-adc/>
>    - $65 HiFiBerry DAC+ ADC Pro
>    <https://www.hifiberry.com/shop/boards/hifiberry-dac-adc-pro/>
>
> Pi options:
>
>    - $12 SD Card SanDisk 64GB Extreme
>    <https://www.amazon.com/dp/B09X7C7LL1>
>    - Pi5:
>       - $60 Pi 5 (4 GB)
>       <https://www.mouser.com/ProductDetail/Raspberry-Pi/SC1111?qs=HoCaDK9Nz5fnLhlMNnKTiQ%3D%3D&src=raspberrypi>
>       / $80 Pi 5 (8 GB)
>       <https://www.mouser.com/ProductDetail/Raspberry-Pi/SC1112?qs=HoCaDK9Nz5c86n0i5EQ%2FPA%3D%3D>
>       - $12 Pi5 PSU
>       <https://www.mouser.com/ProductDetail/Raspberry-Pi/SC1158?qs=HoCaDK9Nz5cK620ydyaGmA%3D%3D&src=raspberrypi>
>    - Pi4:
>       - $35 Pi 4 (1 GB)
>       <https://www.mouser.com/ProductDetail/Raspberry-Pi/SC01929?qs=T%252BzbugeAwjg1Zb0G7gMcFw%3D%3D&src=raspberrypi>
>       / $45 Pi 4 (2 GB)
> [... 25 lines cut ...]
>    outside world is going to need a big switch.
>    - Power: the Pi 5 required 5A @5v <https://github.com/5v>, the Pi 4
>    requires 3A. I'm sure they won't be pulling that much on average but that
>    still means up to around 5V*5A*16 = 400W, plus the power strips for it
>    all.

Oddly with the pi 0 you could use the same wifi router you are using for
the rest of the setup and power the whole thing off of a couple of powered
USB hubs.