micro-nova/AmpliPi #643: Detect communication failures and attempt graceful HW reset

linknum23, 2024-02-14:
### Context
The AmpliPi application running on the Pi does not have direct control over the zone inputs and volumes. It uses an I2C interface to send commands to firmware running on the preamp board. The firmware running on the preamp board controls the zone volumes and inputs over a separate I2C bus. 

### Problem
A customer has a firmware communication issue where the AmpliPi preamp board stops responding to zone/volume control commands and needs to be reset. This happens to them daily.

### Solution
Add a counter for detecting this failure case X times in a short period. When the failure case is detected attempt to reset the firmware and configure it back to its most recent state

rtertiaer, 2024-04-22:
Is this a duplicate of #539?

rtertiaer, 2024-04-22:
Support ticket 116 would have benefited from this.

linknum23, 2024-04-23:
These 2 issues should be combined.