micro-nova/AmpliPi #652: Revisit our HomeAssistant Integration

rtertiaer, 2024-03-04:
We have a good initial rendition of a HomeAssistant integration donated by our community. We should probably revisit this integration and make it more officially supported.

This ticket is a bit of a placeholder for some of that work. Some ideas are sketched out [here](https://github.com/micro-nova/AmpliPi/issues/184); other needs exist around continued testing, making it robust, etc.

linknum23, 2024-03-04:
- Playback uses our announcement feature, it should really allocate and use a source
- zones and groups of zones have entities but how do you use them? It is not immediately obvious
- test how to use this in LMS mode, several users have reported this is very useful
- we need to add this as a pluigin to HACS (and later HA itself). Right now the documentation says were a HACS plugin but it really needs to be manually installed
- the underlying AmpliPi client,  pyamplipi,  probably needs some small updates for changes in our API

rtertiaer, 2024-05-23:
* can homeassistant do dynamic grouping of zones?

rtertiaer, 2024-05-24:
some helpful links:
* https://github.com/brianhealey/hacs_amplipi
* https://github.com/brianhealey/pyamplipi

@mjustian will take a look at this plugin, assess how it could be improved, and start these tasks.

linknum23, 2024-05-28:
- For better integration with amplipi we may need to maintain a fork of this with extra amplipi functionality: https://github.com/kalkih/mini-media-player
- The media player has a good readme on how to install the development version of the media player. Our plugin requires similar steps right now.

rtertiaer, 2024-08-09:
I think it's reasonable to call this complete now, yeah?

mjustian, 2024-08-09:
Yeah

closed 2024-08-09 by rtertiaer