micro-nova/AmpliPi #669: 0.3.4 Update "Fails"

datagitlies, 2024-03-18:
I ran it twice.  Both times it reached this step:

    `Enable linger for pi user`

When it failed

linknum23, 2024-03-18:
Oh no sorry you are having trouble. It sounds like the amplipi webserver is failing to start up for some reason. We need a little more precise logging. Can you send us the output of http://amplipi.local:19531/entries? This will give us a much more fine grained system log.

datagitlies, 2024-03-18:
@linknum23 the good news is that the system seems to be functioning normally.  So the update "fails" but I'm not really sure what happened.  I was able to use the system after the update.

Below are the log entries you requested:

<details>
<summary>entries</summary>

```
Mar 17 20:52:25 amplipi kernel: Booting Linux on physical CPU 0x0
Mar 17 20:52:25 amplipi kernel: Linux version 5.10.103-v7+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1529 SMP Tue Mar 8 12:21:37 GMT 2022
Mar 17 20:52:25 amplipi kernel: CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
Mar 17 20:52:25 amplipi kernel: CPU: div instructions available: patching division code
Mar 17 20:52:25 amplipi kernel: CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Mar 17 20:52:25 amplipi kernel: OF: fdt: Machine model: Raspberry Pi Compute Module 3 Plus Rev 1.0
Mar 17 20:52:25 amplipi kernel: random: fast init done
Mar 17 20:52:25 amplipi kernel: Memory policy: Data cache writealloc
Mar 17 20:52:25 amplipi kernel: Reserved memory: created CMA memory pool at 0x3a000000, size 64 MiB
Mar 17 20:52:25 amplipi kernel: OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
Mar 17 20:52:25 amplipi kernel: Zone ranges:
Mar 17 20:52:25 amplipi kernel:   DMA      [mem 0x0000000000000000-0x000000003dffffff]
Mar 17 20:52:25 amplipi kernel:   Normal   empty
Mar 17 20:52:25 amplipi kernel: Movable zone start for each node
Mar 17 20:52:25 amplipi kernel: Early memory node ranges
Mar 17 20:52:25 amplipi kernel:   node   0: [mem 0x0000000000000000-0x000000003dffffff]
Mar 17 20:52:25 amplipi kernel: Initmem setup node 0 [mem 0x0000000000000000-0x000000003dffffff]
Mar 17 20:52:25 amplipi kernel: On node 0 totalpages: 253952
Mar 17 20:52:25 amplipi kernel:   DMA zone: 2232 pages used for memmap
Mar 17 20:52:25 amplipi kernel:   DMA zone: 0 pages reserved
Mar 17 20:52:25 amplipi kernel:   DMA zone: 253952 pages, LIFO batch:63
Mar 17 20:52:25 amplipi kernel: percpu: Embedded 20 pages/cpu s50828 r8192 d22900 u81920
Mar 17 20:52:25 amplipi kernel: pcpu-alloc: s50828 r8192 d22900 u81920 alloc=20*4096
Mar 17 20:52:25 amplipi kernel: pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
Mar 17 20:52:25 amplipi kernel: Built 1 zonelists, mobility grouping on.  Total pages: 251720
Mar 17 20:52:25 amplipi kernel: Kernel command line: coherent_pool=1M snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:76:15:F8 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=tty1 root=PARTUUID=4d81a4c5-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
Mar 17 20:52:25 amplipi kernel: Kernel parameter elevator= does not have any effect anymore.
                                Please use sysfs to set IO scheduler for individual devices.
Mar 17 20:52:25 amplipi kernel: Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
Mar 17 20:52:25 amplipi kernel: Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
Mar 17 20:52:25 amplipi kernel: mem auto-init: stack:off, heap alloc:off, heap free:off
Mar 17 20:52:25 amplipi kernel: Memory: 923464K/1015808K available (10240K kernel code, 1312K rwdata, 2952K rodata, 1024K init, 862K bss, 26808K reserved, 65536K cma-reserved)
Mar 17 20:52:25 amplipi kernel: SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
Mar 17 20:52:25 amplipi kernel: ftrace: allocating 32081 entries in 95 pages
Mar 17 20:52:25 amplipi kernel: ftrace: allocated 94 pages with 5 groups
Mar 17 20:52:25 amplipi kernel: rcu: Hierarchical RCU implementation.
Mar 17 20:52:25 amplipi kernel:         Rude variant of Tasks RCU enabled.
Mar 17 20:52:25 amplipi kernel:         Tracing variant of Tasks RCU enabled.
Mar 17 20:52:25 amplipi kernel: rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
Mar 17 20:52:25 amplipi kernel: NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
Mar 17 20:52:25 amplipi kernel: random: get_random_bytes called from start_kernel+0x3ac/0x580 with crng_init=1
Mar 17 20:52:25 amplipi kernel: arch_timer: cp15 timer(s) running at 19.20MHz (phys).
Mar 17 20:52:25 amplipi kernel: clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
Mar 17 20:52:25 amplipi kernel: sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
Mar 17 20:52:25 amplipi kernel: Switching to timer-based delay loop, resolution 52ns
Mar 17 20:52:25 amplipi kernel: Console: colour dummy device 80x30
Mar 17 20:52:25 amplipi kernel: printk: console [tty1] enabled
Mar 17 20:52:25 amplipi kernel: Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
Mar 17 20:52:25 amplipi kernel: pid_max: default: 32768 minimum: 301
Mar 17 20:52:25 amplipi kernel: LSM: Security Framework initializing
Mar 17 20:52:25 amplipi kernel: Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
Mar 17 20:52:25 amplipi kernel: Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
Mar 17 20:52:25 amplipi kernel: cgroup: Disabling memory control group subsystem
Mar 17 20:52:25 amplipi kernel: CPU: Testing write buffer coherency: ok
Mar 17 20:52:25 amplipi kernel: CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[... cut ...]
Mar 18 05:54:13 amplipi authbind[835]: alsaloop: pcm.c:1654: snd_pcm_poll_descriptors_count: Assertion `pcm' failed.
Mar 18 05:54:13 amplipi authbind[835]: Subprocess exited with non-zero code: -6. Restarting...
Mar 18 05:55:02 amplipi CRON[1918]: pam_unix(cron:session): session opened for user pi by (uid=0)
Mar 18 05:55:02 amplipi CRON[1919]: (pi) CMD (/home/pi/amplipi-dev/scripts/check-online)
Mar 18 05:55:02 amplipi CRON[1918]: pam_unix(cron:session): session closed for user pi
Mar 18 05:57:56 amplipi authbind[835]: alsaloop: pcm.c:16

datagitlies, 2024-05-25:
@linknum23 I just installed `0.3.5` and received the same error.  

### Update Log
```
09:46:45: installation failed
09:46:44: Enable linger for pi user
09:46:44: Removed amplipi-updater-test.service
09:46:44: Remove amplipi-updater-test.service
09:46:43: Stop amplipi-updater-test.service
09:46:43: active
09:46:43: Check amplipi-updater-test.service status
09:46:43: Enable amplipi-updater-test.service
09:46:42:
Ok!
09:46:42: Check url http://0.0.0.0:5002/update
09:46:42: active
09:46:42: Check amplipi-updater-test.service status
09:46:41: Start amplipi-updater-test.service
09:46:39: Reload systemd config
09:46:38: Created amplipi-updater-test.service
09:46:38: Create amplipi-updater-test.service
09:46:38: Directory /home/pi/.config/systemd/user already exists
09:46:38: Create directory /home/pi/.config/systemd/user
09:46:37: Reload systemd config
09:46:37: Created amplipi-updater.service
09:46:37: Create amplipi-updater.service
09:46:37: Directory /home/pi/.config/systemd/user already exists
09:46:37: Create directory /home/pi/.config/systemd/user
09:46:36: Failed checking version
09:46:36: Checking version reported by API
09:46:36: Enable amplipi.service
09:46:36: Failed to check url, this happens when the server is offline
09:46:35: Check url http://0.0.0.0
09:46:35: active
09:46:35: Check amplipi.service status
09:46:35: Start amplipi.service
09:46:14: Reload systemd config
09:46:14: Created amplipi.service
09:46:14: Create amplipi.service
09:46:14: Directory /home/pi/.config/systemd/user already exists
09:46:13: Create directory /home/pi/.config/systemd/user
09:46:13: Setup autobind
09:46:13: Stop amplipi.service
09:46:12: active
09:46:12: Check amplipi.service status
```

Linked below are the log entries from http://amplipi.local:19531/entries

Attachment: [entries.log](https://github.com/micro-nova/AmpliPi/files/15443990/entries.log)

datagitlies, 2024-07-21:
I just installed version `0.4.0` and the install was successful.  I didn't see any errors during the `Enable linger for pi user` step so I'm assuming this issue is resolved now!

Thus, I'm closing the issue.

closed 2024-07-21 by datagitlies