micro-nova/AmpliPi #676: Default Passwords Too Long for E-Ink Display

SteveMicroNova, 2024-03-28:
[image]
See image, the character limit for default passwords is at least one off to fit on the new display

SteveMicroNova, 2024-03-28:
This may be an issue with this the E-Ink or it may be an issue with how this specific board was mounted, we're also looking into better QA on E-Ink mounting as a solution to this issue as the image looks like the board is shifted about a character worth of space to the right

SteveMicroNova, 2024-03-28:
#677 is related

rtertiaer, 2024-04-19:
I think this is actually just hte mounting on this particular box being off a smidge and am inclined to close this.

rtertiaer, 2024-04-19:
#677 obsoletes this ticket; closing.

closed 2024-04-19 by rtertiaer