micro-nova/AmpliPi #702: LMS stream choppy and distorted after not being used for a few days

sumnerboy12, 2024-05-01:
I have an LMS client stream configured on my AmpliPi, connected to my LMS server running elsewhere on my network. After a few days of no use (with nothing playing and no zones linked) when I try to play something on the LMS stream, the playback is distorted, crackly, and sounds terrible.

Deleting the input using the LMS client stream and recreating makes no difference, the playback is still terrible.

Deleting the input and stream and recreating both appears to resolve the issue.

rtertiaer, 2024-05-29:
I last touched this chunk of code [in Dec 2023](https://github.com/micro-nova/AmpliPi/commit/05f0f01f8f7a91ea4d2ce2fe24226e2a2833b6d4), part of our initial 0.3.0 release. To note - that's also the moment I introduced squeezelite as an actual .deb package, which may have updated the version we are running to one with a regression.

 @sumnerboy12 have you noticed this in other 0.3.x releases as well?

sumnerboy12, 2024-05-29:
I have implemented a nightly restart of all my streams (using NodeRED) and I have not noticed any issues since then. Sorry I can't provide more information but I am happy things are stable at the moment :).

rtertiaer, 2024-06-10:
Reproduced this in-house after running an LMS stream for 6 days. Some notes:
* there are not duplicate `squeezelite` clients
* Modifying AmpliPi controlled volume or mute does not change this.
* One `squeezelite`+LMS player combo was playing Beyonce for 6 days and had this issue. Killing the Beyonce playlist from within LMS and playing a Detroit jazz internet radio station actually cleared this for that player alone; `squeezelite` did not get restarted.

linknum23, 2024-06-10:
One solution to this problem may be to build the latest pianobar release from source. Currently the deb package is at least several years out of date.

rtertiaer, 2024-06-10:
I'm testing upgrades to these packages in #743 ; we'll see if they make a difference.

rtertiaer, 2024-06-24:
We've received some feedback that this issue has not yet reoccurred for an LMS user running 0.3.6, with the above LMS & squeezelite updates merged.

rtertiaer, 2024-06-26:
And this issue has re-occurred for this user.
Given those old suspicious squeezelite issues above, I am increasingly wondering if there's some underlying squeezelite/LMS dependency  issue. All those reports are many years old and have seemingly evaporated - nobody complains of this anymore, except our users. This may be solved long term by #471 + #472 + upgrading to a later distro (which unlocks #415). 

In the shorter term, I think I'm going to implement a cronjob that jams the reset button on the stream every two-ish days, and put a comment next to it that references this comment. Future person: have we upgraded beyond Buster for our distribution? you may be able to get rid of this cronjob!

PR #791 (2024-07-03): Restart LMS clients every day
### What does this change intend to accomplish?
A fast follow to #779 , restart squeezelite clients more frequently. Continues to fix #702 . This is a request from a customer who uses LMS heavily.

### Checklist
None apply; the CHANGELOG language stays valid from #779.
closed 2024-07-02 by rtertiaer