micro-nova/AmpliPi #705: Reset pandora station after disconnecting stream

SteveMicroNova, 2024-05-03:
#692 added support for station browsing within pandora, this issue is a request that the station be reset to default when you close the stream, so you don't disjoint the station from the stream name (see image for example)

I closed this stream and reopened it after using the new functionality from #692 and it is not a classical station anymore, this could be perceived as a bug by those who did not change the station

[image]

linknum23, 2024-05-03:
In addition to this we may want to update the pandora stream settings page
to have a "use current station" when the stream is active.Or better yet
have a right click -> set as default station in the station list.

gorski123, 2024-05-15:
I dont think this is a bug. #692 changed the way a pandora stream works. Its no longer a stream but rather an account. I would argue that the fix here is really that the 'stream' should be renamed to something that reflects the account rather than the station the account is currently playing.

rtertiaer, 2024-05-15:
This might be reasonably imagined as some way to reset a stream if it enters a defunct state. That may look like a button labelled "reset".

Beyond that, a bit of a reimaginging of the long-running streams concept might be a path forward too - what are the user stories for a long running stream? Probably not for this ticket.

SteveMicroNova, 2024-09-26:
Pandora streams are now realized as pandora account instances and not as individual pandora stations, this issue is closed

closed 2024-09-26 by SteveMicroNova