micro-nova/AmpliPi #706: Be more gentle with testing external services

rtertiaer, 2024-05-06:
We have bumped up against IP bans and rate limits when running AmpliPi tests that interact with external services. We should be more gentle with these. Some potential paths forward:
* Segment these tests away from other tests
* Only run these tests on releases or merges to main (this won't fix IP limiting, but would for account limiting)
* Make these tests hit the external services less intensely

The service we encountered this the most with is Pandora, but we really should be gentle with all external services in this way.

linknum23, 2024-05-21:
This happened again today

linknum23, 2024-05-21:
It breaks my pandora client and makes it so I have to load like 500+MB of a chrome tab :sob:

rtertiaer, 2024-05-21:
#730 is one PR that can help address this, tho it's unclear if it will completely close this issue.

closed 2024-05-21 by linknum23

linknum23, 2024-05-28:
We will reopen this if necessary.

commit f78da3fc (2024-05-21): Tests: Limit the amount of requests sent to pandora api

Closes #706