micro-nova/AmpliPi #741: System log persistence during important events

rtertiaer, 2024-06-07:
We do not save the system log right now, largely to save on disk io. We should have a mechanism to save the system log during important events though - when a user requests support, during system upgrades, etc.

rtertiaer, 2024-07-22:
It would also be excellent to select the actual log level, or at least turn on debug logging.

SteveMicroNova, 2024-07-22:
> It would also be excellent to select the actual log level, or at least turn on debug logging.

Is this a request to add a separate setting for the log level on the frontend, to set the default log level to debug, or to ensure that the log level is debug when the boolean is true?

rtertiaer, 2024-07-22:
This is a request to either add:
* a drop down to select the log level
* or a boolean to turn on DEBUG logging

I had to swap the log level in `lms_metadata.py` to [something less verbose than DEBUG already](https://github.com/micro-nova/AmpliPi/pull/802) because it was absolutely inundating and wiping out the system log - but it would be nice to be able to turn that back on when necessary.