micro-nova/AmpliPi #766: Constant "Error: 'playlist_loop', trying again in 2 seconds..." messages when LMS playlist is empty

sumnerboy12, 2024-06-23:
I am running LMS on a different server (192.168.10.112). I noticed recently that if I clear the playlist on LMS, for the AmpliPi client I have configred (via the LMS web UI) I start see the following messages in the Amplipi logs;

```
"2024-06-23T22:49:05.868+12:00","amplipi","DEBUG:urllib3.connectionpool:Starting new HTTP connection (1): 192.168.10.112:9000"
"2024-06-23T22:49:05.872+12:00","amplipi","DEBUG:urllib3.connectionpool:http://192.168.10.112:9000 ""POST /jsonrpc.js HTTP/1.1"" 200 240"
"2024-06-23T22:49:05.876+12:00","amplipi","Error: 'playlist_loop', trying again in 2 seconds..."
"2024-06-23T22:49:05.877+12:00","amplipi","INFO:__main__:Error: 'playlist_loop', trying again in 2 seconds..."
"2024-06-23T22:49:06.685+12:00","amplipi","INFO:     192.168.10.21:59830 - ""GET /api HTTP/1.1"" 200 OK"
"2024-06-23T22:49:07.885+12:00","amplipi","DEBUG:urllib3.connectionpool:Starting new HTTP connection (1): 192.168.10.112:9000"
"2024-06-23T22:49:07.889+12:00","amplipi","DEBUG:urllib3.connectionpool:http://192.168.10.112:9000 ""POST /jsonrpc.js HTTP/1.1"" 200 239"
"2024-06-23T22:49:07.893+12:00","amplipi","Error: 'playlist_loop', trying again in 2 seconds..."
"2024-06-23T22:49:07.893+12:00","amplipi","INFO:__main__:Error: 'playlist_loop', trying again in 2 seconds..."
```

As soon as I add something to the LMS playlist these message stop, and then immediately start again if I clear the playlist. Seems very reproducible.

rtertiaer, 2024-06-24:
Some of this is definitely as a result of setting the log level to DEBUG; it's exposed urllib3's DEBUG  level logging as well, which is definitely less than ideal.

sumnerboy12, 2024-06-24:
Yeah not too worried about the `DEBUG` logging, although agree it would be nice to quieten it down a bit.

It is the `Error:` logging that concerns me - is that an error being returned from LMS or generated in the AmpliPi LMS client code?

sumnerboy12, 2024-06-24:
Ah, just had a look at the changes in #768 - looks like you guys are all over it :).

closed 2024-06-27 by SteveMicroNova