micro-nova/AmpliPi #777: Stream input fields do not match font-family for the rest of the app

rtertiaer, 2024-06-27:
[image]

This changed recently, and while the fields look wayyyy better than the old variety, it still does not look good.

SteveMicroNova, 2024-06-28:
I imagine this was caused during [PasswordSecurity](https://github.com/micro-nova/AmpliPi/pull/716/files), specifically when I replaced our bespoke TextField with the Material UI variant

To chase this breadcrumb trail, use [the docs for the component](https://mui.com/material-ui/react-text-field/) to see how to control the font, I'm sure you can add a className but if that doesn't work the sx prop should work (it's like the style prop but passes stuff down to the children instead)

I self assign this, but will be out for 10 business days. Follow this breadcrumb trail if you feel it's worth solving while I'm gone; otherwise consider this comment a note to self for me.

rtertiaer, 2024-07-10:
Closed with #800

closed 2024-07-10 by rtertiaer