micro-nova/AmpliPi #786: Make a hosted API for release management

rtertiaer, 2024-06-28:
This will permit us to release versions based on hw revision, create beta channels, do canary deploys, etc.

rtertiaer, 2024-12-05:
How I imagined this working:
a) an AmpliPi POSTs an object that contains its current version, its firmware version, and perhaps some hardware details (eink? eeprom?)
b) an API serves up a list of objects that describes all compatible releases, including md5sums and possibly GPG signatures from micronova-keyring keys. It may describe both the tarball, and full disk image.

This allows us to gate releases - you can only upgrade to AmpliPi 7 if you upgrade to 6 first, not directly from 5. This also helps us ensure whatever download we get is not corrupted.

linknum23, 2024-12-05:
A second generation of this would also probably include what release stream
the customer was in sable, nightly, ...