micro-nova/AmpliPi #809: Button color

SteveMicroNova, 2024-07-11:
Somewhere between now and the 0.4.0 release there's been a change to main that changes the color of all buttons grey
See before and after below

Before:
[image]
[image]

After:
[image]
[image]

SteveMicroNova, 2024-07-11:
Note that these buttons aren't inactive, simply discolored

rtertiaer, 2024-07-11:
This is as a result of #800 ; if you'd like, you can modify the theme for this ticket. The blue that it was using prior was not stylistically consistent with the rest of the things, but "grey" is the only theme my colorblind self could pick that was sane-ish.

linknum23, 2024-07-12:
I would vote for buttons being having some color. Grey buttons look like they are disabled.

closed 2024-07-17 by SteveMicroNova