micro-nova/AmpliPi #810: FM Radio only outputs to left speaker

SteveMicroNova, 2024-07-11:
https://amplipi.discourse.group/t/fm-radio-is-in-mono/310/4

Found by a discourse user, tested in house, FM radio is transmitted in Mono but for some reason it only goes to the left speaker rather than both

linknum23, 2024-07-11:
For reference this was previously fixed here:
https://github.com/micro-nova/AmpliPi/pull/455. We should figure out what
got broken.

Stuff like this really points at us needing a human in the loop play all of
the basic stream types script, with the human validating that we get good
audio and metadata for each.

rtertiaer, 2024-07-12:
I was not able to replicate this on my appliance.

SteveMicroNova, 2024-07-15:
> I was not able to replicate this on my appliance, playing [101.9 WDET](https://wdet.org/) 😎

Interesting, the inhouse unit did replicate this with both [100.3 WNIC](https://wnic.iheart.com/) and [94.7 WCSX](https://wcsx.com/)
What differences can we draw between our units? Is yours a streamer? Is your usb dongle somehow better? Is it something about the speakers, using RCA out instead of the pheonix connectors maybe?

SteveMicroNova, 2024-07-15:
I've walked around the office to check both units, here's my findings:

> Is yours a streamer?
No

> Is your usb dongle somehow better?
No, it is in fact the exact same. Not even just the same model, the same antenna itself.

> Is it something about the speakers, using RCA out instead of the pheonix connectors maybe?
This is my bet, the inhouse only uses pheonix (amplified) and your test unit uses RCA out (preout) connections

Pedalnetics, 2024-07-16:
I am using the Phoenix connectors and getting Mono audio (I am the original poster in the forums who hit this issue).

rtertiaer, 2024-07-26:
our new QA process has replicated this as well

klay2000, 2024-07-30:
this appears to only happen on source 0

closed 2024-08-05 by klay2000