micro-nova/AmpliPi #848: Internet Radio Streams Disappeared and Can't Be Added After Update

boulwarek, 2024-07-31:
I installed version 0.4.0 and my internet radio streams are no longer visible.
When trying to add the same streams back, I receive an error "create stream failed: expected string or bytes-like object"
I did not look what version I was previously on, but these same streams worked fine before this update.

An example URL of one of the streams: https://us9.maindigitalstream.com/ssl/knza

rtertiaer, 2024-07-31:
Thanks lots for your report! I'm investigating now.

I can reproduce this on my 0.4.1 test box. The full traceback:

```
Jul 31 12:57:20 amplipi authbind[821]: ERROR:amplipi.ctrl:Traceback (most recent call last):
Jul 31 12:57:20 amplipi authbind[821]:   File "./amplipi/ctrl.py", line 1031, in create_stream
Jul 31 12:57:20 amplipi authbind[821]:     stream = amplipi.streams.build_stream(data, mock=self._mock_streams)
Jul 31 12:57:20 amplipi authbind[821]:   File "./amplipi/streams.py", line 1855, in build_stream
Jul 31 12:57:20 amplipi authbind[821]:     return InternetRadio(name, args['url'], args.get('logo'), disabled=disabled, mock=mock)
Jul 31 12:57:20 amplipi authbind[821]:   File "./amplipi/streams.py", line 1134, in __init__
Jul 31 12:57:20 amplipi authbind[821]:     self.validate_stream(url=self.url, logo=self.logo)
Jul 31 12:57:20 amplipi authbind[821]:   File "./amplipi/streams.py", line 1255, in validate_stream
Jul 31 12:57:20 amplipi authbind[821]:     if 'logo' in kwargs and not re.fullmatch(URL_LIKE, kwargs['logo']):
Jul 31 12:57:20 amplipi authbind[821]:   File "/usr/lib/python3.7/re.py", line 178, in fullmatch
Jul 31 12:57:20 amplipi authbind[821]:     return _compile(pattern, flags).fullmatch(string)
Jul 31 12:57:20 amplipi authbind[821]: TypeError: expected string or bytes-like object
```

closed 2024-08-01 by rtertiaer

rtertiaer, 2024-08-01:
hey @boulwarek ! I've merged a PR that will fix this bug and it will be present in the 0.4.2 release, which I'm pressing today or tomorrow. (If you wanted to get this bug fix super early and feel up to the task, you can also run our shell script `./scripts/deploy pi@amplipi.local` on your local machine from a checkout of  this repo's `main` branch.)

In look at this, it seems like the streams got explicitly discarded from the configuration due to them failing validation :/ This is pretty painful and we don't take that lightly; please accept our apologies. As a result, we blocked upgrades to all affected versions until this bugfix is released. I'm also going to immediately work on taking regular dated backups of system configs in issue #850 so we can _always_ recover from a similar incident. 

During upgrades and system reboots, we take a backup of the system configuration, either at `/home/pi/amplipi-dev/house.json.bak` or `/home/pi/.config/amplipi/house.json.bak`. It might be worthwhile to take a look at the revision of that file to see if it contains the missing streams, or you can email us at support@micro-nova.com to have a look through a support tunnel.

boulwarek, 2024-08-01:
Thanks for the super quick response to this! I reviewed the house.json.bak file and it does not contain the radio streams that vanished. This might be due to the fact that I once I saw your issue mentioning the logo being required, I re-added a couple of stations with logos to get them running. This isn't a big deal in my case. I only have a handful of streams and can manually add them back.

PR #849 (2024-08-01): Fix internet radio logo field validation bug
### What does this change intend to accomplish?
This PR fixes #848 . The `logo` field is marked as `Optional[str]` in the function that calls `validate_sttream`, and a `None` gets passed in when we do not specify one. This broke the regex processing; I've gated that with a truthy test first.

I also took the opportunity to revisit a couple of stream's `validate_stream()` methods to explicitly ensure required values are set. We don't bump into this because our frontend also does field validation for required fields, but helps the API/automation case. [... cut]