micro-nova/AmpliPi #892: Individual items in the WebUI's settings dialog no longer present as links

rtertiaer, 2024-08-16:
Hovering over individual items does not change my cursor to indicate these things are links as opposed to just a list. This regressed sometime between https://github.com/micro-nova/AmpliPi/commit/c4b280ac451e2beeb1d560f3cb5da1341861b63b and right now's HEAD, https://github.com/micro-nova/AmpliPi/commit/9880b29c26347d053988cfd9d8f9a9550c01a541. There were 3 commits between then and there to `/web/src`:
* https://github.com/micro-nova/AmpliPi/commit/e15e761d21180f8a9cb83e5fd6a349a9eb16f605
* https://github.com/micro-nova/AmpliPi/commit/a698ec6d684de573459e241f98d336ffd4b8f899
* https://github.com/micro-nova/AmpliPi/commit/bc0b6cc040447aa02fa0c3acf16d7857bc592d92

klay2000, 2024-08-19:
This also appears to affect the onhover effect so this is probably a CSS mistake somewhere.

rtertiaer, 2024-08-19:
`git bisect` tells me dd8d3a3b6cb05cca72ebb16c6580d7b9d7ba0028 is the bad commit

PR #897 (2024-08-19): Make clickable ListItems -> ListItemButtons; migrate more to MUI
### What does this change intend to accomplish?
This PR fixes #892 , by using `ListItemButton`s instead of `ListItem`s. It also moves some ugly `input`s to mui `TextField`s. Here are some before and after screenshots:

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Multiple browser & viewport tests are gonna happen concurrent to this PR (ie, those last two bullet points.)

### Checklist

* [x] Have you tested your changes and ensured they work?
* [x] Have you checked to ensure there aren't other ope
closed 2024-08-19 by rtertiaer