micro-nova/AmpliPi #898: Stream selection header in settings truncated in small iOS

rtertiaer, 2024-08-19:
[image]

The device is an iPhone 12 Mini running iOS 17.5.1.

rtertiaer, 2024-08-19:
This is actually true for all of the individual submenus in settings, not just streams.

SteveMicroNova, 2024-08-19:
This seems to be Issue #554, solved in #767

rtertiaer, 2024-08-19:
I was testing off main's `HEAD`, commit 2eb34a1170eca267fed6868aa4175e664c4796ad; it would seem that PR was an incomplete fix.

SteveMicroNova, 2024-08-22:
Taking this on, going to start writing notes in here as I investigate

First note is that this is fixed in the app, but not in web browsers
It is worse on mobile chrome than mobile safari

App:
[image]

Safari:
[image]

Chrome:
[image]

rtertiaer, 2024-08-22:
fwiw chrome on ios is actually just a reskinned safari with extra spying

Lohrer, 2024-08-22:
> Mobile firefox is completely fine

It's cut off on my mobile firefox.

SteveMicroNova, 2024-08-22:
> It's cut off on my mobile firefox.

🥴

closed 2024-08-23 by rtertiaer