micro-nova/AmpliPi #907: Stream selection modal font size too small

SteveMicroNova, 2024-08-23:
I don't currently know if it was intentional as it __does__ make long stream names fit better, but the font size of the stream selection modal is currently `1rem` instead of `1.5rem`. See attached pics:

Before:
[image]

After:
[image]

SteveMicroNova, 2024-08-23:
Seems it's from a mui package updating, the current state of `main` is uploaded to inhouse and doesn't have it, yet the current state of main on my computer does

rtertiaer, 2024-08-23:
This is a result of #901 converting the items in this list to an MUI `List`, which applies the typography/theming stuff from `App.jsx`. It's not just the font size - it's also the font itself, which should be Raleway but that typography style specifies as Open Sans

closed 2024-10-21 by SteveMicroNova