micro-nova/AmpliPi #910: Allow for different asound.conf files

gorski123, 2024-08-26:
We previously equalized and limited DAC ouputs in #213. The DAC limit was driven by the dynamic range of the opamps on =< rev 3.A preamp boards. The current rev (>=4.A) preamp boards have modified front end circuitry to support up to 2Vrms audio signals. We currently can take advantage of that through the RCA inputs, but the DAC level is still limited. The DAC output level is set in asound.conf. This file auto loads on startup. We should add a mechanism that allows us to load a differet asound.conf file based on the detected hardware. Note that we are also currently loading the same asound.conf files on the streamer boxes, which artificially limits the DAC output levels.

linknum23, 2024-08-26:
This may not be needed if we implement #911

gorski123, 2024-08-26:
yup, dropping this.

closed 2024-08-26 by gorski123