micro-nova/AmpliPi #934: Mobile icon alignment

rickballard, 2024-09-21:
The AmpliPi logo is super tiny and slightly misaligned, in the folder drawer of the Android 14 home screen.  Likely it needs a quick fix by someone of more esteemed rank than I. 
[image]

linknum23, 2024-09-22:
That is quite small and a tad low. Let's see what we can do about cleaning that up.

Thanks Rick!

rtertiaer, 2024-11-15:
To add some data to this, it appears the icon does show up as a square, with some help from the launcher:

[image]

rickballard, 2024-11-15:
Native launcher, Pixel 9 Pro XL, latest version Android 15.