micro-nova/AmpliPi #976: LMS Mode toggle downloads a config in both directions

SteveMicroNova, 2024-10-14:
Downloading a config of an LMS mode enabled amplipi is useless as it's basically only ever gonna be lms 1-4 connected to zones 1-4

rtertiaer, 2024-10-14:
I readded this functionality because there's nothing locking down folks from modifying their config and then toggling back, thus wiping out their config. I'd either:
a) lock down LMS mode so you can't change your config, or
b) leave this as-is

Doing neither and taking out this funcitonality creates problems.

rtertiaer, 2024-10-14:
https://github.com/micro-nova/AmpliPi/pull/805

SteveMicroNova, 2024-10-14:
Locking down LMS mode is a larger undertaking than I am interested in at the moment and so will remove the changes from my #974 branch
Once we get to the point of taking on that issue, this one can be solved via removing `ConfigDownload()` from the `onApply` portion of the `lmsMode=true` path of `LMSMode.jsx` in the settings directory

closed 2024-10-14 by rtertiaer